gnuradio/gnuradio #1632: QPSK constellation diagram and (Doxygen) comment mismatch

m3x1m0m, 2018-02-22:
There is a mismatch between the comment for QPSK constellation

https://github.com/gnuradio/gnuradio/blob/master/gr-digital/include/gnuradio/digital/constellation.h

and how is is actually assigned 

https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/constellation.cc

iff little endian format is assumed.

noc0lour, 2018-02-25:
Since I suggested posting an issue here I'll add some of my thoughts: 
 - The block is treating incoming bytes with little endianness (`01` -> 2), which is not documented
 - Little endian is unintuitive to the common user
 - Having an option to switch endianness might be useful, though I don't know if this is too much overhead and documentation is enough. (Actually, a GREP regarding endianness for blocks with byte-in/byte-out ports would be useful)

marcusmueller, 2018-03-02:
GREP or no grep, we need to document the current behaviour. @m3x1m0m, due you feel up to the task of fixing those comments?

m3x1m0m, 2018-03-04:
@marcusmueller Sure!

PR #1690 (2018-03-22): Better documentation explaining what QPSK is doing.
Fixes #1632.
commit b27079c9 (2018-03-22): Better documentation explaining what QPSK is doing.

Fixes #1632.
closed 2018-03-22 by marcusmueller