gnuradio/gnuradio #1634: Replace PyQwt with PyQtGraph

noc0lour, 2018-02-23:
PyQwt is effectively unmaintained since 2014 and thus has no support for Qt5/Qwt6 and Python3. 
Fortunately there is PyQtGraph which could be used to replace all occurences of PyQwt.

Occurences:  
 - [ ] gr-filter/python/filter/design/filter_design.py
 - [ ] gr-filter/python/filter/gui/idealbanditems.py
 - [ ] gr-filter/python/filter/gui/pyqt_filter_stacked.py
 - [ ] gr-filter/python/filter/gui/polezero_plot.py
 - [ ] gr-qtgui/apps/plot_form.py
 - [ ] gr-qtgui/apps/plot_time_form.py
 - [ ] gr-utils/python/utils/pyqt_filter.py
 - [ ] gr-utils/python/utils/pyqt_plot.py
 - [ ] gr-zeromq/examples/python/gui.py

ekigwana, 2018-06-05:
Any progress on this? It'd be nice to be able to use the filter design tool.

noc0lour, 2018-06-06:
I barely looked into it, any help is appreciated.

ekigwana, 2018-06-07:
I have started to look at this and having only PyQt5 I have noticed that a few other changes will be required. From PyQt5  [gotchas](http://pyqt.sourceforge.net/Docs/PyQt5/gotchas.html) I noticed QStrings are gone and PyQt5 is not compatible with PyQt5. I forked the next branch and I believe it is targeting PyQT5 and Python v2, is that correct? Is PyQt4 support also required?

noc0lour, 2018-06-07:
If possible I'd recommend to have development in py2/py3 compat mode. With GNU Radio 3.8 the goal is to support py3 as well.

ekigwana, 2018-06-09:
To keep things reasonable since QT4 is EOL and I don't even have a system with it, I am targeting the following:

1. Python v2/v3
2. PyQt5
3. pyqtgraphics

ekigwana, 2018-06-12:
I have been working on the filter design tool and have a question below.

New "Filter Taps" plot in progress unless it is not required
[image]
Old "Filter Taps" plot
[image]
New "Impulse Response" plot
[image]
Old "Impulse Response" plot
[image]
New "Step Response" plot
[image]
Old "Step Response" plot
[image]

What is the purpose of the "Filter Taps" plot? It is exactly the same as the "Impulse Response" but with a different x axis label and different tab label. It looks pretty useless to me as an Electrical engineer and think it should go. I am going some space between the view border and the axis labels so not done yet.

marcusmueller, 2018-08-24:
@ekigwana somehow this slipped my attention. Is your code somewhere that I can look at it and merge it?

noc0lour, 2018-10-01:
I tried the `pyqtgraph` branch on the @ekigwana remote, I actually rebased that on the new master. Maybe that helps with springstarting this development again?

ekigwana, 2018-10-08:
Sorry I got too busy. From what I could tell it should be "fairly" easy.

btashton, 2018-10-19:
@noc0lour @ekigwana Anything I can do to help here?

noc0lour, 2018-11-04:
@btashton Sure, if you wish you can pick my branch and open a separate PR for tracking/discussing your progress on that matter. Other than @marcusmueller rejecting `pyqtgraph` as dependency I'd say go for it.

btashton, 2018-11-04:
@marcusmueller mentioned here #2145 that pyqtgraph might be up for discussion.

btashton, 2018-11-04:
@marcusmueller @noc0lour before I look into picking this up too much, are we going to have a CLA issue, I don't want to spend too much time and then get blocked by needing a CLA from @ekigwana

noc0lour, 2018-11-04:
Good question. My gut feeling tells me we might need one from @ekigwana . @bhilburn ?

ekigwana, 2018-11-04:
I really just ran out of time to work on this but I'll give you whatever you need

ekigwana, 2018-11-04:
Also when I looked at the filter design code, it looked like someone was transitioning to a modern interface so there is otherwise quite a bit of dead code sitting in there. I like the look of https://github.com/chipmuenk/pyFDA but it uses matplotlib although the concept still works a bit better for that. For this to work, the issues raised in #2145 would certainly need to be addressed in a project wide manner. My two cents FWIW.

bhilburn, 2018-11-05:
Yes, we'll need a CLA from @ekigwana. Can you e-mail me? bhilburn@gnuradio.org

ekigwana, 2018-12-19:
edwardwwgk@gmail.com

btashton, 2018-12-19:
Sweet, I'll pick the rest of the filter stuff up once this is done.  The plotting stuff I think is in a good place, just need to move the code over and add the file seek interface to the GUI.

bhilburn, 2018-12-20:
@ekigwana - Just sent you an e-mail with instructions for getting the CLA in place!