gnuradio/gnuradio #1636: gr-blocks: bad function for determining total number of tags passed through tag_debug() block

mrjacobagilbert, 2018-02-24:
Function: https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/tag_debug_impl.cc#L67

At some point tags go out of scope and get cleaned up. Quick python FG to demonstrate:

```
from gnuradio import gr, blocks
import pmt
tb = gr.top_block()
tag = gr.tag_utils.python_to_tag((0, pmt.intern("tag"), pmt.PMT_NIL, pmt.PMT_NIL))
src = blocks.vector_source_c(range(1000), True, 1, [tag,])
head = blocks.head(gr.sizeof_gr_complex*1,100000)
tag_dbg = blocks.tag_debug(gr.sizeof_gr_complex*1, '', "");
tag_dbg.set_display(False)
tb.connect(src, head, tag_dbg)
tb.run ()
print "should have gotten 100 tags, got " + repr(tag_dbg.num_tags())
```

This also does not expose information for other inputs.

marcusmueller, 2018-03-02:
oh, shoot, yeah, that's true, and it's bad on a software-architectural level (lack of ownership clarity).

Pinging @fwunsch here, too: Would that happen to the messages stored in a message_debug, too?

@mrjacobagilbert, do you agree: with the current model, we should probably be doing deep copies of the tags themselves in tag_debug?

mrjacobagilbert, 2020-08-31:
@marcusmueller going through old issues...

yes, i agree deep copies should address this at some potential performance cost, but this is a debug block so IMO faithful execution of stated objectives is more important.

marcusmueller, 2023-12-21:
Should probably be fine with current architecture an pybind11, but someone needs to check!