gnuradio/gnuradio #1637: next fails to build on Ubuntu 16.04

jdemel, 2018-02-24:
I tried to build 'next' on my Ubuntu 16.04 machine and it fails. I have GCC 5.4 and SWIG 2.0.12 on that machine.

First, CMAKE tells me these components are enabled:

```
-- ######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * python-support
--   * testing-support
--   * volk
--   * doxygen
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-audio
--   * * alsa
--   * * oss
--   * gr-channels
--   * gr-trellis
--   * gr-utils
--   * gr-vocoder
--   * gr-wavelet
--   * gr-zeromq
--   * gnuradio-companion
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * sphinx
--   * gr-comedi
--   * gr-qtgui
--   * gr-uhd
--   * gr-video-sdl
```
First thing I wondered, why are `gr-comedi` and `gr-video-sdl` still around? As fas as I can tell, they were never enabled whenever I installed GR3.7. It seems like they always miss a dependency.

Next, I run `make`. Essentially the build fails with
```
[ 11%] Built target _runtime_swig_swig_tag
[ 11%] Linking CXX executable runtime_swig_gnuradio_runtime_swig_7dd5e
Swig source
[path]/gnuradio-runtime/swig/hier_block2.i:37: Error: Syntax error in input(1).
gnuradio-runtime/swig/CMakeFiles/runtime_swig_gnuradio_runtime_swig_7dd5e.dir/build.make:207: recipe for target 'gnuradio-runtime/swig/runtime_swig_gnuradio_runtime_swig_7dd5e' failed
make[2]: *** [gnuradio-runtime/swig/runtime_swig_gnuradio_runtime_swig_7dd5e] Error 1
make[2]: *** Deleting file 'gnuradio-runtime/swig/runtime_swig_gnuradio_runtime_swig_7dd5e'
CMakeFiles/Makefile2:1288: recipe for target 'gnuradio-runtime/swig/CMakeFiles/runtime_swig_gnuradio_runtime_swig_7dd5e.dir/all' failed
make[1]: *** [gnuradio-runtime/swig/CMakeFiles/runtime_swig_gnuradio_runtime_swig_7dd5e.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```
The problematic line is:
'hier_block2.i' line 37: `noexcept(false);`
I tried moving back to the way these files look on 'master' and I tried commenting them out. But these are quite a lot of uses and also, it feels like `noexcept(false)` is the way to go here and SWIG should deal with it.

Does SWIG miss any special command? How can I compile 'next'?

noc0lour, 2018-02-25:
Looking at the CI build for the latest commit on `next` in a Ubuntu 16.04 container: https://ci.gnuradio.org/buildbot/#/builders/6/builds/425 even testsand build with `all` options enabled passed.

Your swig version is rather old though. At least for the vanilla Ubuntu 16.04 container swig is already on `3.0.8`. The last time we checked dependencies for an upcoming GNU Radio 3.8 release we actually specified swig 2.0.11 because Centos 7 and Ubuntu 14.04 are still shipping those. If we want to support both (and swig 2.0.11) we should definitely this with those in the CI system. If we bump build time dependencies to swig 3.x.x because everyone else is shipping it and we don't want to depend on a five year old swig I'd advise to upgrade to swig 3.x.x.

jdemel, 2018-02-26:
Updating SWIG to 3.0.8 did the trick. I wonder how such an old version ended up on my machine.

Since CentOS and Ubuntu 14.04 both ship 2.0.11 'next' should compile with this SWIG version. Unless, GR3.8 would be deferred until 2019. In that case 14.04 support could be dropped. But I really hope GR3.8 will be released earlier this year.

The GR CI system seems to be quite fragile, but I guess it would be great to check if 'next' builds on 14.04 in the future. Also, I guess @noc0lour, you're the guy to make that happen. It's great that GR has that CI system.

Since the original issue is fixed, we could close this issue. But some questions remain:
- How to ensure Ubuntu 14.04/CentOS7 support?
- Why does the build fail with old SWIG?

marcusmueller, 2018-03-02:
I agree, let's close this issue.

As closing remarks (hurr, durr):

* We do not care about Ubuntu 14.04 on 3.8. 
* We (I) do care about CentOS 7 support, but it's actually not that absurd to demand non-demand installation of SWIG from whoever tries to *build* for CentOS, imho; at this point, the technological debt we carry by depending on old SWIG is simply too large; note that there's third party repos where you can get swig 3.0.12 binaries for CentOS 7 from.
* because old SWIG is old and as you put it "should, but does not, deal with it".

closed 2018-03-02 by marcusmueller