gnuradio/gnuradio #1638: gr-uhd: master branch fails to build (using uhd master branch)

mrjacobagilbert, 2018-02-25:
Master branch of gnuradio fails to build against the master branch of UHD. Build works if either GR or UHD are built off maint. It is also possible this is a UHD issue.

Errors:

```
[ 88%] Building CXX object gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o
In file included from /usr/local/include/uhd/usrp/dboard_iface.hpp:16:0,
                 from /usr/local/include/uhd/usrp/multi_usrp.hpp:36,
                 from /home/jacob/sdr/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_block.h:28,
                 from /home/jacob/sdr/gnuradio/gr-uhd/lib/usrp_block_impl.h:26,
                 from /home/jacob/sdr/gnuradio/gr-uhd/lib/usrp_block_impl.cc:23:
/usr/local/include/uhd/usrp/gpio_defs.hpp:77:5: error: in C++98 ‘uhd::usrp::gpio_atr::gpio_attr_map’ must be initialized by constructor, not by ‘{...}’
     };
     ^
/usr/local/include/uhd/usrp/gpio_defs.hpp:77:5: error: could not convert ‘{{GPIO_SRC, uhd::usrp::gpio_atr::GPIO_ATTR_SRC}, {GPIO_CTRL, uhd::usrp::gpio_atr::GPIO_ATTR_CTRL}, {GPIO_DDR, uhd::usrp::gpio_atr::GPIO_ATTR_DDR}, {GPIO_OUT, uhd::usrp::gpio_atr::GPIO_ATTR_OUT}, {GPIO_ATR_0X, uhd::usrp::gpio_atr::GPIO_ATTR_ATR0X}, {GPIO_ATR_RX, uhd::usrp::gpio_atr::GPIO_ATTR_ATRRX}, {GPIO_ATR_TX, uhd::usrp::gpio_atr::GPIO_ATTR_ATRTX}, {GPIO_ATR_XX, uhd::usrp::gpio_atr::GPIO_ATTR_ATRXX}, {GPIO_READBACK, uhd::usrp::gpio_atr::GPIO_ATTR_READBACK}}’ from ‘<brace-enclosed initializer list>’ to ‘const gpio_attr_map_t {aka const std::map<uhd::usrp::gpio_atr::gpio_attr_t, std::__cxx11::basic_string<char> >}’
/usr/local/include/uhd/usrp/gpio_defs.hpp:79:70: error: ‘>>’ should be ‘> >’ within a nested template argument list
    static const std::map<gpio_attr_t, std::map<uint32_t, std::string>> attr_val
                                                                     ^

<SNIP>

/usr/local/include/uhd/usrp/gpio_defs.hpp:127:5: error: could not convert ‘{{uhd::usrp::gpio_atr::GPIO_ATTR_CTRL, uhd::usrp::gpio_atr::gpio_ctrl_mode}, {uhd::usrp::gpio_atr::GPIO_ATTR_DDR, uhd::usrp::gpio_atr::gpio_direction}, {uhd::usrp::gpio_atr::GPIO_ATTR_OUT, uhd::usrp::gpio_atr::gpio_level_map}, {uhd::usrp::gpio_atr::GPIO_ATTR_ATR0X, uhd::usrp::gpio_atr::gpio_level_map}, {uhd::usrp::gpio_atr::GPIO_ATTR_ATRRX, uhd::usrp::gpio_atr::gpio_level_map}, {uhd::usrp::gpio_atr::GPIO_ATTR_ATRTX, uhd::usrp::gpio_atr::gpio_level_map}, {uhd::usrp::gpio_atr::GPIO_ATTR_ATRXX, uhd::usrp::gpio_atr::gpio_level_map}, {uhd::usrp::gpio_atr::GPIO_ATTR_READBACK, uhd::usrp::gpio_atr::gpio_level_map}}’ from ‘<brace-enclosed initializer list>’ to ‘const std::map<std::__cxx11::basic_string<char>, std::map<std::__cxx11::basic_string<char>, unsigned int> >’
gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/build.make:62: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o' failed
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o] Error 1
CMakeFiles/Makefile2:14429: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all' failed
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

noc0lour, 2018-02-25:
Looks like C++11 syntax might crept into public UHD headers. (Which I thought should be C++98 compatible?). If your log4cpp5 isn't new enough the GNU Radio build will not use C++11 thus failing on the UHD `gpio_defs` header.

mrjacobagilbert, 2018-02-25:
Interesting. This was on an updated Ubuntu 16.04 (log4cpp5 1.0-4.1 installed). This isn't holding me up since I am ok w/ UHD maint branch, but I'm pretty sure a basic pybombs install will fail right now.

dkozel, 2018-02-25:
@mrjacobagilbert This issue is in UHD and was also found internally at Ettus. We have a fix committed which will be pushed to the public branch very soon and before the upcoming release.
http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2018-February/027734.html

mbr0wn, 2018-02-28:
It's public -- I tried building against latest master. Thanks @mrjacobagilbert!

marcusmueller, 2018-03-02:
Sorry, reopening this, just for clarification: @mbr0wn, users have the unfortunate habit of sticking to versions with very particular specialities – can you at least pinpoint the commit that resolved this issue, and better even, the first one that introduced it? (I presume you wouldn't have closed this if you thought the lifespan of that bug was long enough to warrant a GR-side workaround, so that's fine with me)

dkozel, 2018-03-04:
@marcusmueller 
I believe it is:
Introduced on Feb 21
https://github.com/EttusResearch/uhd/commit/21cc92016c4d5c3ae99e8fb49c04793feb51c21e

Fixed Feb 23
https://github.com/EttusResearch/uhd/commit/594030c1af4af36f5590df58f71606c62d74f033

marcusmueller, 2018-03-04:
Thank you! This closes the issue.

closed 2018-03-04 by marcusmueller

TarikKazaz, 2020-10-02:
Dear,

 I am having the same issue with installing RFNOC following the guide from https://kb.ettus.com/Getting_Started_with_RFNoC_Development. 

I performing the installation on Ubuntu 18.04.5 LTS.
And I would like to keep UHD 4.0.

My error log is the same as above:

gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/build.make:86: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o' failed
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o] Error 1
CMakeFiles/Makefile2:14726: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all' failed
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gnuradio:
Build failed.
[ERROR] Error installing package gnuradio. Aborting.

Does anyone have any hint for a solution?

Thanks in advance,

Tarik

panizprs, 2021-01-05:
Dear TarikKazaz,

Have you found any solution for it? I got the same error as you and I don't know what I should do?

hemanthR-aiila, 2022-09-07:
> Dear,
> 
> I am having a similar issue with installing RFNOC following the guide from https://kb.ettus.com/Getting_Started_with_RFNoC_Development.
> 
> I performing the installation on Ubuntu 18.04.5 LTS. And I would like to keep UHD 4.0.
> 
> My error log is the similar as above:
> 
> ```
> [ 86%] Building CXX object gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o
> In file included from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/rpcregisterhelpers.h:26:0,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/basic_block.h:42,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/block.h:27,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/sync_block.h:27,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_block.h:26,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_block_impl.h:26,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.h:23,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:24:
> /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/rpcmanager.h:56:17: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
>      static std::auto_ptr<rpcserver_booter_base> boot;
>                  ^~~~~~~~
> In file included from /usr/include/c++/7/bits/locale_conv.h:41:0,
>                  from /usr/include/c++/7/locale:43,
>                  from /usr/include/boost/format.hpp:23,
>                  from /home/tkazaz/rfnoc/include/uhd/types/dict.ipp:12,
>                  from /home/tkazaz/rfnoc/include/uhd/types/dict.hpp:154,
>                  from /home/tkazaz/rfnoc/include/uhd/types/device_addr.hpp:11,
>                  from /home/tkazaz/rfnoc/include/uhd/stream.hpp:11,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/gr_uhd_common.h:26,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:23:
> /usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
>    template<typename> class auto_ptr;
>                             ^~~~~~~~
> In file included from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/rpcregisterhelpers.h:26:0,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/basic_block.h:42,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/block.h:27,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/sync_block.h:27,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_block.h:26,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_block_impl.h:26,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.h:23,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:24:
> /home/tkazaz/rfnoc/src/gnuradio/gnuradio-runtime/include/gnuradio/rpcmanager.h:57:17: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
>      static std::auto_ptr<rpcserver_booter_aggregator> aggregator;
>                  ^~~~~~~~
> In file included from /usr/include/c++/7/bits/locale_conv.h:41:0,
>                  from /usr/include/c++/7/locale:43,
>                  from /usr/include/boost/format.hpp:23,
>                  from /home/tkazaz/rfnoc/include/uhd/types/dict.ipp:12,
>                  from /home/tkazaz/rfnoc/include/uhd/types/dict.hpp:154,
>                  from /home/tkazaz/rfnoc/include/uhd/types/device_addr.hpp:11,
>                  from /home/tkazaz/rfnoc/include/uhd/stream.hpp:11,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/gr_uhd_common.h:26,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:23:
> /usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
>    template<typename> class auto_ptr;
>                             ^~~~~~~~
> In file included from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.h:24:0,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:24:
> /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:31:19: error: redefinition of ‘struct uhd::stream_args_t’
>  struct GR_UHD_API stream_args_t {
>                    ^~~~~~~~~~~~~
> In file included from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/gr_uhd_common.h:26:0,
>                  from /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:23:
> /home/tkazaz/rfnoc/include/uhd/stream.hpp:58:16: note: previous definition of ‘struct uhd::stream_args_t’
>  struct UHD_API stream_args_t
>                 ^~~~~~~~~~~~~
> /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In constructor ‘gr::uhd::usrp_source_impl::usrp_source_impl(const uhd::device_addr_t&, const uhd::stream_args_t&, bool)’:
> /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:74:7: error: class ‘gr::uhd::usrp_source_impl’ does not have any field named ‘_recv_timeout’
>        _recv_timeout(0.1), // seconds
>        ^~~~~~~~~~~~~
[... 32 lines cut ...]
> /home/tkazaz/rfnoc/src/gnuradio/gr-uhd/lib/usrp_source_impl.cc:608:41: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
[... cut ...]
> CMakeFiles/Makefile2:14726: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all' failed
> make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
> Makefile:162: recipe for target 'all' failed
> make: *** [all] Error 2
> [ERROR] Build failed. See output above for error messages.
> [ERROR] Problem occurred while building package gnuradio:
> Build failed.
> [ERROR] Error installing package gnuradio. Aborting.
> ```
> 
> Does anyone have any hint for a solution?
> 
> Thanks in advance,
> 
> Tarik

Hi , i am also getting same problem you got any solution ?  please inform me

hemanthR-aiila, 2022-09-07:
> @marcusmueller I believe it is: Introduced on Feb 21 [EttusResearch/uhd@21cc920](https://github.com/EttusResearch/uhd/commit/21cc92016c4d5c3ae99e8fb49c04793feb51c21e)
> 
> Fixed Feb 23 [EttusResearch/uhd@594030c](https://github.com/EttusResearch/uhd/commit/594030c1af4af36f5590df58f71606c62d74f033)

 Hi , i am also getting same error can u please tell how to solve it  
[image]

markibonge, 2023-09-11:
I am also having similar issues: Please see log below 

In file included from /home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.h:24:0,
                 from /home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:24:
/home/opnear/workarea/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:31:19: error: redefinition of ‘struct uhd::stream_args_t’
 struct GR_UHD_API stream_args_t {
                   ^~~~~~~~~~~~~
In file included from /home/opnear/workarea/gnuradio/gr-uhd/lib/gr_uhd_common.h:26:0,
                 from /home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:23:
/usr/local/include/uhd/stream.hpp:47:16: note: previous definition of ‘struct uhd::stream_args_t’
 struct UHD_API stream_args_t
                ^~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In constructor ‘gr::uhd::usrp_source_impl::usrp_source_impl(const uhd::device_addr_t&, const uhd::stream_args_t&, bool)’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:74:7: error: class ‘gr::uhd::usrp_source_impl’ does not have any field named ‘_recv_timeout’
       _recv_timeout(0.1), // seconds
       ^~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:75:7: error: class ‘gr::uhd::usrp_source_impl’ does not have any field named ‘_recv_one_packet’
       _recv_one_packet(true),
       ^~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘virtual void gr::uhd::usrp_source_impl::set_recv_timeout(double, bool)’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:464:5: error: ‘_recv_timeout’ was not declared in this scope
     _recv_timeout = timeout;
     ^~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:464:5: note: suggested alternative: ‘set_recv_timeout’
     _recv_timeout = timeout;
     ^~~~~~~~~~~~~
     set_recv_timeout
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:465:5: error: ‘_recv_one_packet’ was not declared in this scope
     _recv_one_packet = one_packet;
     ^~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:465:5: note: suggested alternative: ‘one_packet’
     _recv_one_packet = one_packet;
     ^~~~~~~~~~~~~~~~
     one_packet
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘void gr::uhd::usrp_source_impl::flush()’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:512:29: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
         _dev->get_device()->recv(outputs,
                             ^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:516:49: error: ‘RECV_MODE_FULL_BUFF’ is not a member of ‘uhd::device’
                                  ::uhd::device::RECV_MODE_FULL_BUFF,
                                                 ^~~~~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘virtual int gr::uhd::usrp_source_impl::work(int, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:600:44: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
     size_t num_samps = _dev->get_device()->recv(output_items,
                                            ^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:604:64: error: ‘RECV_MODE_FULL_BUFF’ is not a member of ‘uhd::device’
                                                 ::uhd::device::RECV_MODE_FULL_BUFF,
                                                                ^~~~~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:608:41: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
         num_samps = _dev->get_device()->recv(output_items,
                                         ^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:612:61: error: ‘RECV_MODE_ONE_PACKET’ is not a member of ‘uhd::device’
                                              ::uhd::device::RECV_MODE_ONE_PACKET,
                                                             ^~~~~~~~~~~~~~~~~~~~
gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/build.make:86: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o' failed
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o] Error 1
CMakeFiles/Makefile2:14760: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all' failed
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
Makefile:162: recipe for target 'all' failed

marcusmueller, 2023-09-12:
Hi, everyone. "I get the same error" is definitely not helping. If you have a different problem (like Tarik, markibonge might), please open a *new* issue here. Don't post "similar" issues under existing ones. I'm cleaning the issue comments here and lock this issue.