gnuradio/gnuradio #1641: Failure to install via pybombs due to <windows.h>?

moppymopperson, 2018-02-26:
About 2~3 weeks ago I had GNRadio up and running with pybombs, but during now I'm trying to install it again and getting errors that I did not last time. It's failing during CMake, not during actual building or linking.

I'll paste the full output below this, but I want to put the contents of `gnuradio/build/CMakeFiles/CMakeError.log` first, since it's particularly suspicious. What would cause CMake to think this is running on windows? I'm on Ubuntu Artsy, which I realize isn't officially supported, but I didn't have any issues just a week or two ago.

```
/home/erik/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx:1:10: fatal error: windows.h: No such file or directory
 #include <windows.h>
          ^~~~~~~~~~~
compilation terminated.
CMakeFiles/cmTC_f1dbf.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_f1dbf.dir/CheckIncludeFile.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_f1dbf.dir/CheckIncludeFile.cxx.o] Error 1
make[1]: Leaving directory '/home/erik/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_f1dbf/fast' failed
make: *** [cmTC_f1dbf/fast] Error 2
```

Full output

```
erik@erik-workstation  ~  pybombs -v install gnuradio
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - Parsing config file failed (/etc/pybombs/config.yml).
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/erik/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: CommentedMap([('default_prefix', '/home/erik/pybombs')])
PyBOMBS.ConfigManager - DEBUG - Template directory: /home/erik/miniconda3/lib/python3.6/site-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/erik/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/erik/pybombs)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/erik/pybombs/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/erik/pybombs/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/erik/pybombs/src
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /home/erik/pybombs/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/erik/pybombs/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/erik/pybombs/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/home/erik/pybombs/.pybombs/recipes', '/home/erik/.pybombs/recipes/gr-recipes', '/home/erik/.pybombs/recipes/gr-etcetera', '/home/erik/miniconda3/lib/python3.6/site-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git', 'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/erik/pybombs/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/erik/pybombs/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/erik/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/erik/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 178 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/erik/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/erik/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 97 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/erik/miniconda3/lib/python3.6/site-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/erik/miniconda3/lib/python3.6/site-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.install - DEBUG - Initializing command class for command install
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.PackageManager - DEBUG - apt is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager zypper
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pymod
PyBOMBS.PackageManager - DEBUG - pymod is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
[... 288 lines cut ...]
-- Python checking for numpy - not found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency CHEETAH_FOUND = TRUE
--   Dependency LXML_FOUND = TRUE
--   Dependency PYGTK_FOUND = TRUE
--   Dependency NUMPY_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:45 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/erik/pybombs/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/erik/pybombs/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```

noc0lour, 2018-02-26:
apparently numpy is missing, thus cmake fails because gnuradio-companion is force-enabled.

moppymopperson, 2018-02-26:
That was my initial interpretation too, but numpy is installed, so it must be either unable to find numpy for some reason, or a different error is manifesting itself as a numpy missing error.

moppymopperson, 2018-02-27:
Got it pinned. It wasn't playing nicely with Anaconda. Pybombs executed the check for numpy with sudo, which caused it use the version of python in `/usr/bin` instead of the `~/miniconda3/bin`. The version of python packaged with Ubuntu does indeed not include numpy, which lead to the error. 

The solution in my case was to install pip through apt in addition to Anaconda, then install numpy using `sudo pip install numpy` to install numpy to the OS included version of python instead of the Anaconda one.

It's building now - hopefully there won't be any more hitches.

closed 2018-02-27 by moppymopperson