gnuradio/gnuradio #1642: gr-grpc support

reiinakano, 2018-02-26:
Any plans to include gr-grpc? Kind of like gr-zeromq? Would you be interested in a PR adding that in here?

noc0lour, 2018-02-26:
@reiinakano It's something to think about going forward. Maybe not for the 3.7.x  series since grpc would be a new dependency.

warmonkey, 2018-04-08:
well in 3.7.x thrift actually depends on specific version 0.9.3, that is not included in ubuntu 16.04 lts release. and thrift is really hard to use.

reiinakano, 2018-04-11:
> we're evaluating how to add more RPC functionality

@marcusmueller, actually, my company is working on an open source project called starcoder https://github.com/infostellarinc/starcoder. It's not close to being ready for a public release, but we built it so we can dynamically start, stop, and send/receive data to/from flowgraphs in any language of choice through gRPC (we use Go). We're in the process of migrating our internal code to use it, and once it's stable enough, maybe release publicly. We want it to be a good option for deploying flowgraphs in prod.

devnulling, 2020-01-21:
@reiinakano For this to be added in tree, it really needs to go through the GREP process. Please submit a GREP here for more discussion:

https://github.com/gnuradio/greps/blob/master/grep-0000-greps.md

NeuroForLunch, 2021-06-24:
Can gr-grpc be an OOT module?

mormj, 2021-10-04:
Typically a new in-tree module would start as someone's OOT that is proven to work well and show some value for being upstreamed.  Then as @devnulling said, it would go through the GREP process to get agreement that it should go in tree.  Closing this as a GR issue, since at this point it would need to start as a solid OOT module.

closed 2021-10-04 by mormj