gnuradio/gnuradio #1643: Compile .grc files to .py files programmatically?

reiinakano, 2018-02-27:
Sorry if this is documented somewhere, but I couldn't find it. Is there a way to convert .grc files to .py files programmatically or at the very least through the command line?

reiinakano, 2018-02-27:
Saw grcc. Sorry!

closed 2018-02-27 by reiinakano

noc0lour, 2018-02-27:
No probem, if there is a way to improve documentation so someone else will find this information faster, please elaborate :)

reiinakano, 2018-02-27:
Well it was buried here https://wiki.gnuradio.org/index.php/Hackfest1211#GRCC

In fact I spent about an hour digging through grc source code to find how to do this until I searched for https://github.com/gnuradio/gnuradio/search?utf8=%E2%9C%93&q=Platform%28&type= and found the `utils.grcc` file.

Would be good to add a sidenote here: https://wiki.gnuradio.org/index.php/GNURadioCompanion