gnuradio/gnuradio #1644: Replace matplotlib with pyqtgraph

noc0lour, 2018-02-27:
Due to being a lot faster and PyQwt has to be replaced by PyQtGraph anyway the same can be done with matplotlib.

A comprehensive list of utils using matplotlib or pylab:
- [ ] gr-digital/examples/snr_estimators.py
- [ ] gr-digital/examples/example_costas.py
- [ ] gr-digital/examples/berawgn.py
- [ ] gr-digital/examples/example_timing.py
- [ ] gr-digital/examples/example_fll.py
- [ ] gr-digital/python/digital/test_soft_decisions.py
- [ ] gr-filter/examples/synth_to_chan.py
- [ ] gr-filter/examples/channelize.py
- [ ] gr-filter/examples/interpolate.py
- [ ] gr-filter/examples/resampler.py
- [ ] gr-filter/examples/reconstruction.py
- [ ] gr-filter/examples/fir_filter_ccc.py
- [ ] gr-filter/examples/chirp_channelize.py
- [ ] gr-filter/examples/fir_filter_fff.py
- [ ] gr-filter/examples/fft_filter_ccc.py
- [ ] gr-filter/examples/decimate.py
- [ ] gr-filter/examples/synth_filter.py
- [ ] gr-dtv/examples/atsc_ctrlport_monitor.py
- [ ] gnuradio-runtime/examples/volk_benchmark/volk_plot.py
- [ ] gr-fec/python/fec/polar/channel_construction_bec.py
- [ ] gr-fec/python/fec/polar/channel_construction.py
- [ ] gr-fec/python/fec/polar/channel_construction_awgn.py
- [ ] gr-fec/python/fec/polar/testbed.py
- [ ] gr-analog/examples/fmtest.py

jdemel, 2018-03-02:
This issue is related to #1634.

According to [PyQtGraph](http://www.pyqtgraph.org/), `PyQtGraph` is different in a couple of ways, when compared to `matplotlib`.
- `PyQtGraph` is not as feature rich as `matplotlib`. Whenever you want to use offline tools, this is probably a big feature. If you want online tools, you might prefer speed though.
- It seems like the `python-pyqtgraph` package is only available for Ubuntu 16.04 and newer. Thus, this will cause problems for Ubuntu 14.04. And maybe other older distros.
- According to [PyQtGraph](http://www.pyqtgraph.org/), `PyQt4` might be used. But it also works with `PySide` (for Qt5). I assume for GR, the `PySide` implementation is desired. Could this cause any issues? The `PyQtGraph` docs state that `PyQt4` is prefered.

noc0lour, 2018-03-02:
Thanks for raising the discussion: 
 - I still think we should not target a new release for a Linux distribution which will most likely not include new software (as LTS distributions should do)
 - PyQtGraph can be used with PyQt4, PySide or PyQt5. In fact I used it in a PyQt5-only context. [Changelog](https://github.com/pyqtgraph/pyqtgraph/blob/develop/CHANGELOG)

Facing the issue that we can't use `PyQwt` going forward and have to replace it with another plotting library (which we could do by reimplementing plotting tools in C++ using Qwt).
Also matplotlib is used almost exclusively in examples, by cutting matplotlib out of dependencies this might have benefits.

marcusmueller, 2018-03-02:
I fully second @noc0lour's sentiment here. We simply don't support anachronistic concepts like the existence of the ether, or support for Ubuntu 14.04, with release 3.8.

The qt4 statement from the pyqtgraph docs is worth investigating, but I honestly think it's obsolete. I only tested my last patches against that with qt5 myself.