gnuradio/gnuradio #1647: too many arguments to function volk_32f_8u_polarbutterfly_32f

Red-Owl, 2018-02-27:
[ 39%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_common.cc.o
/usr/src/gnuradio/gr-fec/lib/polar_decoder_common.cc: In member function ‘void gr::fec::code::polar_decoder_common::butterfly_volk(float*, unsigned char*, int, int, int)’:
/usr/src/gnuradio/gr-fec/lib/polar_decoder_common.cc:128:95: error: too many arguments to function
         volk_32f_8u_polarbutterfly_32f(llrs, u, block_size(), block_power(), stage, u_num, row);

polar_decoder_common.cc
128:95

noc0lour, 2018-02-27:
Will be fixed by #1640 . You can apply that patch locally if you like. Thanks for reporting :)

noc0lour, 2018-03-10:
This was fixed with 00c6f258259c61f25dc220402a9d07544942b414 on next. If `volk` is released for the `3.7.12` release the outstanding PR will be merged into master.

closed 2018-03-10 by noc0lour

marcusmueller, 2018-03-27:
@Red-Owl just for records: Now fixed on master, and will be fixed in release 3.7.12

pswiatki, 2018-03-29:
Interestingly I now get:

[...]/gnuradio/gr-fec/lib/polar_decoder_common.cc: In member function 'void gr::fec::code::polar_decoder_common::butterfly_volk(float*, unsigned char*, int, int, int)':
[...]/gnuradio/gr-fec/lib/polar_decoder_common.cc:128:81: error: too few arguments to function
         volk_32f_8u_polarbutterfly_32f(llrs, u, block_power(), stage, u_num, row);

I pulled the latest commits from master.

BeamPower, 2018-05-22:
I'm getting the same error.
Can someone provide details on how to fix it?
Thank you.

noc0lour, 2018-05-22:
Update to the newest release on volk & GNU Radio.

barcating, 2018-12-04:
> This was fixed with [00c6f25](https://github.com/gnuradio/gnuradio/commit/00c6f258259c61f25dc220402a9d07544942b414) on next. If `volk` is released for the `3.7.12` release the outstanding PR will be merged into master.

It works, Thanks.