gnuradio/gnuradio #1649: noise_amp and AWGN process doesn't seem to correspond

WRAR891, 2018-02-28:
The AWGN process doesn't seem to correspond with the usually accepted model. If I chose  noise_amp = 0, the  output signal values from the dynamic_channel_model are all +,- inf or inf+infj. If I chose noise_amp  = inf, the  output signal values from the dynamic_channel_model are significantly attenuated version of the input. This contradicts AWGN intuition. Essential, a small value for noise_amp should return a signal with little added noise where as a large value should return a signal with a lot of added noise. How can I tune noise_amp  to give zero AWGN (i.e. not add noise?). Also, for complex signals, the noise variance has to be scaled by sqrt(2), and applied to both the real and imaginary parts. I'm not sure that this is being handled.

WRAR891, 2018-02-28:
I thought this was posted to dynamic_channel_model, I guess it went to the general gnuradio repo. Anyway, this pertains to dynamic_channel_model

mbr0wn, 2018-02-28:
This is the right place to post things like this.

mbr0wn, 2018-02-28:
I haven't taken a look at the block myself, but I would argue that whatever we do should be well reflected in the documentation. Anything else I would classify as a bug.

marcusmueller, 2023-12-21:
Complicated enough for me to think this will actually require significant debugging