gnuradio/gnuradio #1652: Naming confusion: gr::block::d_setlock is a *mutex*, not a *lock*

marcusmueller, 2018-02-28:
Renaming would be an API change.

mbr0wn, 2018-02-28:
Is it worth it though? I say no.

marcusmueller, 2018-03-01:
Well, I've not learned about the usability of `d_setlock` up until not so long ago, because I thought it was actually a block execution lock. Adding the "documentation" label – though I'm not sure where and how to document this.

Turns  out, I'm not the only one who reinvented a setter mutex in the tree, let alone in OOTs.

This begs the question: what's the benefit of having a mutex in the gr::block motherclass, if only very few blocks use it, and there's no downside to declaring your own private mutex, because your _impl model doesn't allow inheritance from actual block implementations, anyway? Should this be getting the "Cruft" tag instead? 

I did a quick scan of all `gr-*` git repos in gr-recipes, and only gr-ais uses `d_setlock` at a single point – the corr_est_cc that, for some reason, was copied from in-tree into this OOT (probably a slight modification). So, I'd argue, if we break this API, we're very unlikely to break anyone's application, and I'm all about cleaning up GR's heart whenever we can – hence the question whether we break API now that we can!

noc0lour, 2018-03-05:
I second that knowledge about `d_setlock` is not widespread. Talking with @skoslowski I have to agree that having a pure virtual interface class for blocks rather than having some implementation only few people know about is favorable.

I'd vote for removing this member since it is not used and only confusing to have.

closed 2021-03-23 by marcusmueller