gnuradio/gnuradio #1665: Configuring incomplete, errors occurred!

Arinerron, 2018-03-05:
Hey! I tried installing gnuradio in a vm, and i got this error:

```
...
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency CHEETAH_FOUND = TRUE
--   Dependency LXML_FOUND = TRUE
--   Dependency PYGTK_FOUND = TRUE
--   Dependency NUMPY_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:45 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```

So, here's the bottom of that log file:
```
aaron@aaron-virtualbox ~/onionr/exp $ tail /usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log
/usr/bin/c++     -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized    -o CMakeFiles/cmTC_90e44.dir/CheckIncludeFile.cxx.o -c /usr/local/src/gnuradio/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx
/usr/local/src/gnuradio/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx:1:21: fatal error: windows.h: No such file or directory
compilation terminated.
CMakeFiles/cmTC_90e44.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_90e44.dir/CheckIncludeFile.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_90e44.dir/CheckIncludeFile.cxx.o] Error 1
make[1]: Leaving directory '/usr/local/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_90e44/fast' failed
make: *** [cmTC_90e44/fast] Error 2
```

Any ideas?

jdemel, 2018-03-05:
`Dependency NUMPY_FOUND = FALSE`
CMake could not find NumPy. Is it installed? On Ubuntu, the required package is 'python-numpy'.

noc0lour, 2018-03-05:
This looks like, both, a missing dependency and a mishap of PyBOMBS, since it should auto detect missing numpy.

closed 2018-03-05 by noc0lour