gnuradio/gnuradio #1666: GRC embedded Python blocks don't work anymore

argilo, 2018-03-05:
When an embedded Python block is added to a GRC flowgraph, the following error appears:
```
Param - Code(_source_code):
	'py_io_signature' object is not iterable
```
It seems there is no way to resolve the error.

I suspect the issue is related to ba63bd46e3a79d4312a16d9d35f918aa4cfa67db; GRC probably needs to be updated to use the new IO signature format.

@willcode

willcode, 2018-03-05:
Yes, most likely my fault - I haven't used Python blocks from GRC and 
didn't consider that. I'll take a look at it.

marcusmueller, 2018-03-05:
Thanks, @willcode, and thanks @argilo for reporting.

marcusmueller, 2018-03-05:
The culprit seems to be in `epy_block_io.py`'s `_ports()`:

```python
def _ports(sigs, msgs):
    ports = list()
    for i, dtype in enumerate(sigs):
…
```

marcusmueller, 2018-03-05:
@willcode gut feeling, adding a quick 

```python
    def __iter__(self):
        return self
    def next(self):
         for elem in self.__types:
             yield elem
```

to your py_io_signature could help!

marcusmueller, 2018-03-05:
No, wait, I don't like that. That way, an instance can only be iterated through once.

Instead
```python
def __iter__(self):
     return iter(self.__types)
```

would be both simpler and smarter.

willcode, 2018-03-06:
This fix seems to work, until we can find a way to teach GRC about 
Python blocks with a variable number of ports. In the mean time, it will 
assume num i/o is equal to the number of types.

Marcus, since it's just a couple lines, would you like to put this in 
yourself? I'll think about how to make GRC ask for number of 
inputs/outputs correctly when I get some time.

marcusmueller, 2018-03-06:
@willcode I will do that.

marcusmueller, 2018-03-06:
Calling @skoslowski on this, because that's an architectural aspect of GRC, and chances are that he's already considered this, so that @willcode doesn't need to :)

willcode, 2018-03-06:
Good idea. The problem is that GRC usually controls the blocks, using 
xml files for hints, but for the Python block it uses introspection. So, 
no component actually knows how many ports the user wants.

The solution is probably to have GRC detect whether there is a variable 
number of ports (via introspection) just as it would get that info from 
an xml file, then add num inputs/outputs to the dialog.

skoslowski, 2018-03-06:
You are correct. I can have a closer look later this week.
If you want to take a stab at this, let me know.

PR #1673 (2018-03-06): runtime: make py_io_signature iterable, hashable
This allows GRC to iterate the output signatures of epy blocks after
switching over to py_io_signature.

Fixes #1666.
commit 713629cc (2018-03-06): runtime: make py_io_signature iterable, hashable

This allows GRC to iterate the output signatures of epy blocks after
switching over to py_io_signature.

Fixes #1666.
closed 2018-03-06 by marcusmueller