gnuradio/gnuradio #1668: Add convenience wrapper for firdes_window to GRC

noc0lour, 2018-03-05:
Similar to `variable_low_pass_filter_taps.xml` a `variable_window_taps.xml` could be added to generate window taps in GRC without relying on manually importing and figuring out parameters for `firdes_window`.

No code has to be written. Adding a `xml` with correct parameters for dialogs on block creation are sufficient.

supra08, 2018-12-24:
@noc0lour, can I take up this issue?

noc0lour, 2018-12-25:
sure, if you need any help you can ask here or ping me in IRC #gnuradio or in the GNU Radio Slack.

supra08, 2019-01-02:
@noc0lour, can you give me the details of the correct parameters for `variable_window_taps.xml`?

sidkapoor97, 2020-04-20:
@noc0lour aren't the low pass window taps being generated by `variable_low_pass_filter_taps.block.yml` I believe the master branch doesn't have `variable_low_pass_filter_taps.xml` file, it's in maint 3.7 branch