gnuradio/gnuradio #1674: CMake: Be sure we still need all checks we do in 3.8

marcusmueller, 2018-03-06:


jdemel, 2018-03-10:
Did you add the `Release 3.9` milestone by accident? I'd argue unnecessary checks in 3.8 should be removed in 3.8 and thus, this issue should be part of the `Release 3.8` milestone.

marcusmueller, 2018-03-22:
No, this was intentional! There's a truckload of dependency in movement for 3.8, and I'd like to let the dust settle on that before we spend too much time on cleaning up checks; it's a configure-time performance enhancement, only, after all.

Which really doesn't mean that if you see a check we don't need, you shouldn't be removing it right now :)

It's just that I can't assign priority to this right now.

marcusmueller, 2021-03-22:
I think I'm through the thick of this, especially in gr-digital

closed 2021-03-22 by marcusmueller