gnuradio/gnuradio #1675: Improvise check for block(s) removal in modtool_rm.py

swap-nil7, 2018-03-06:
[here](https://github.com/gnuradio/gnuradio/blob/master/gr-utils/python/modtool/modtool_rm.py#L51) the code is accepting the empty spaces as argument but this shouldn't be the case as such block cannot exist.

swap-nil7, 2018-03-06:
I think we can fix it by repacing `len(self._info['pattern']) == 0` with `if(not self._info['pattern'] or self._info['pattern'].isspace())`
@noc0lour can you please give your suggestions?

noc0lour, 2018-03-06:
Probably this is a good idea, to have this a bit more pythonic and to not accept strings only containing space characters.

marcusmueller, 2018-03-06:
I'm a bit confused – this is a github issue, but it reads like you meant to post a pull request?

If you meant to post a pull request, please do so! (You'll have to have a git branch from master, which contains your fix)

commit 6ab59c75 (2018-03-22): modtool: Improve check for block(s) removal

Fixes #1675
closed 2018-03-06 by marcusmueller