gnuradio/gnuradio #1680: msvc: remove cmake/msvc/config.h

noc0lour, 2018-03-10:
For more standard conformance we should strive to remove `cmake/msvc/config.h` as soon as possible.

Most of the things in there should be easy fixable.
 - inline -> __inline typedef for `C89` 
 - `ptrdiff_t` -> `ssize_t` typedef can be removed if all `ssize_t` references are replaced with `std::ptrdiff_t`
 - MSVC < 1800 shouldn't be necessary if we use `C++11` since `C++11` and MSVC only play well together for MSVC >= 1800
 - random()/srandom() calls should be replaced with C++11 `<random>` library instead.

If we fix these things this header will be unecessary.

noc0lour, 2018-03-10:
maybe @guruofquality or @gnieboer can comment if implementing this is will be useful and might result in removing `config.h` statements everywhere?

noc0lour, 2018-03-10:
This somehow supersedes #1593

noc0lour, 2018-03-10:
remark: instead of `#defining` missing types in `config.h` globally for MSVC and using deprecated POSIX layer low-level IO functions can be implemented as wrapper and conditionally compiled using `cmake`. 
https://docs.microsoft.com/en-us/cpp/c-runtime-library/low-level-i-o

Note to self: need some testing environment for that.

gnieboer, 2018-03-11:
I agree all the MSVC < 1800 can be taken out, I don't think we need to have compatibility with those versions of MSVC, VS Community Edition is free so there should be little need for most users to need to use an old version, plus I would say the install base would be tiny.

As for the other changes, when I do the 3.7.12 build, I'll check to see what is really necessary and report back.

marcusmueller, 2018-03-22:
@gnieboer thanks!

noc0lour, 2018-03-22:
To throw a pointer: #1685 was brought up due to being the sole consumer of `ssize_t`.

mbr0wn, 2018-03-23:
We can probably also deprecate C89.

marcusmueller, 2018-03-23:
Phew, that'd be very cool, for VOLK at least (right, @n-west ?). I honestly, however, have no idea about the level of C99 or later support in MSVC – my understanding is that MSVC is a "C++ first" compiler, and C support is lagging more than C++ support.

dkozel, 2018-03-23:
MSVC 2015/2017 has good but not perfect C99 support. Anecdotally I think we're unlikely to hit the limitations in any meaningful way. There are some advanced macro features which are buggy etc.
https://docs.microsoft.com/en-us/cpp/visual-cpp-language-conformance

marcusmueller, 2018-03-23:
I'd be very happy to push for min requirement being MSVC2015 in 3.8. Is there someone specific we should ask about this?

dkozel, 2018-03-23:
I don't believe so. Very very few people are building from source and not using Geof's scripts. When we get to the point that it is straightforward to do so or OOTs can be compiled separately then maintaining support for older versions might become more important. I think you can make the call to drop < 2015.

guruofquality, 2018-03-24:
Second that. I Stopped building PothosSDR for MSVC2013 months and months ago, just due to the effort of building and patching things for two compilers. Now I think theres several packages that require 2015 simply due to its better C++11 support. Theres no going back at this point anyway, so I wont  be complaining :-)

gnieboer, 2018-03-27:
Did a bit of checking, and there are two bits of code needed:

1- define ssize_t as ptrdiff_t... I could posit that really ptrdiff_t is probably the correct application in the code so if everything was refactored this need would go away naturally.  It's in a couple places only, so could also be placed directly with the files affected.

2- define rand() as random()... definitely needed.

As some compiler/MSVC related code is in attributes.h, these *could* be moved there, though I note that file is strictly macro redefinitions.

All else could be removed without any compile errors (at least on MSVC 2015 as previously discussed)

marcusmueller, 2018-03-27:
re 2- define `rand` as `random`: we've (hopefully!) just kicked out `rand` and `rand48` completely, so I think this is no longer an issue, @gnieboer?

re 1- ssize_t -> ptrdiff_t hm, I think that'd be worth a separate issue for tracking, no?

noc0lour, 2018-03-27:
I have some changes for msvc lined up on next. ssize_t typdefing is only taking place in 5 files which I already have in there.

mormj, 2021-03-01:
code from https://github.com/gnuradio/gnuradio/pull/4104 was merged manually on master.

I believe this addresses the main goal of the issue - please reopen if there is more that needs to be addressed.

closed 2021-03-01 by mormj