gnuradio/gnuradio #1681: Cannot declare fixed-size array in embedded Python block

lutusp, 2018-03-10:
This issue was mentioned a year ago with respect to embedded Python blocks, but apparently wasn't fixed:

`gr.sync_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[(np.complex64,4096)],
            out_sig=[(np.complex64,4096)]
        )`

When interpreted, the above produces: "Invalid conversion specification." Is this wrong? How does one declare a fixed-length vector?

willcode, 2018-03-11:
That looks correct, I just tried making a Python block in GRC having the same signature, using the current master branch, and it worked.

Could you provide an example GRC file, reference to the bug report, version information, or other info?

willcode, 2018-03-11:
@lutusp if you are using 3.7.10, then you are using code that is at least a year old. So, no year-old bugs would have been fixed. Have you tried a recent version?

lutusp, 2018-03-11:
@willcode Okay, thanks -- I managed not to notice the version mismatch. I've decided to deal with this without relying on the embedded approach, so people who use my app won't have to upgrade Gnuradio to be able to use it (Ubuntu and others are distributing Gnuradio 3.7.10). I would have moved away from embedded anyway once I had debugged my code.

Thanks for taking a look at this.

marcusmueller, 2018-03-22:
@lutusp, I'll go ahead and close this issue for now, since I can't reproduce it. If **anybody** stumbles across the same problem with the current release of GNU Radio, please reopen!

(and I mean it that way: we're really happy about every bug report we get, but we just overhauled the embedded python io type system, and I think the likelihood of this no longer applying is just very high. I'd be happy if you actually tested this again as soon as we released 3.7.12!)

closed 2018-03-22 by marcusmueller