gnuradio/gnuradio #1685: blocks: deprecate & remove file_descriptor_sink

noc0lour, 2018-03-11:
It would be best if we deprecate this block in 3.7.12 and remove it in 3.8. 

It is written in a highly unportable fashion, which could be replaced by a similar block using ASIO to abstract the file descriptor idiom for different OSes without using incompatible types.

jdemel, 2018-03-18:
I assume that the `file_descriptor_sink` is a very useful block if one wants to interface GR with other applications. I'd suggest to rephrase the title: **make file_descriptor_sink portable**. I'd say that would better match your issue description.

noc0lour, 2018-03-18:
I tend to agree with you, basically if we take in `asio` (note: can be used without boost) we should factor out functionality in `file_sink` to a base "kernel" and just convert .the "file_descriptor|handle" to something usable.

marcusmueller, 2018-03-22:
I like  both your ideas,  so let's keep this issue's title as-is, but keep in our head that if someone needs that  block, we should be having something "up our sleeves" to replace it. 

FYI, it was used to drive the old blks2 TCP blocks, and that was a nightmare. My wild guess is it was written for that purpose, but I don't know.

mbr0wn, 2018-03-23:
It's a nice block to connect to a Unix pipe, which would then go to another application (see also @jdemel's comment).

marcusmueller, 2018-03-23:
Hm, pipes should be able to get a proper file name, but I see that point. So, what to do with this? Point being, there *really* should be no functional difference to a file sink, right? It does the same thing on the same OS interface. Can't we code-deduplicate by making the file_sink as capable as useful, and just have an alternative `make`, or something?

Also note that I can see a lot of room for improvement of the file_sink; e.g. if we also do a "fixed size file sink" (to avoid extraneous Head blocks), there's nothing stopping us from e.g. `mmap`ing files beforehand, and reserving the file on the file system at start rather than extending it during run (which might really make a difference for large files on cluttered FSes, when it comes to maximum sustainable streaming rate to disk for vendors of high-perf super-awesome SDRs).

noc0lour, 2018-03-23:
@marcusmueller I fear reworking the file_sink part with all of it's use cases (e.g. pipes/handles) iis its own "little" project, which also would probably mean using asio. Probably a *very* good idea to take into 3.8

marcusmueller, 2018-03-23:
Yes! I agree. I don't think we've fleshed this idea out enough to split out a separate issue for that, so we'll keep this issue for now.

marcusmueller, 2018-03-24:
@willcode nice example of current usage, so no, won't remove file_descriptor_*.

closed 2018-03-24 by marcusmueller