gnuradio/gnuradio #1686: moving_average_ff blocks actually do a moving sum, name is misleading

MattEttus, 2018-03-13:
The moving average blocks require a scale of 1/N in order to actually compute a moving average.  They are really moving sum blocks.  This is confusing for the user and violates the principle of least surprise.

Suggest either changing the name or default to scaling by 1/N.

noc0lour, 2018-03-15:
This is probably harder to implement than one might think since the scaling is configured alongside with length in the GUI. 
A possible solution might be an option block for "default scaling using `1/length`" enabled by default and "custom scaling". 

Or removing the scale option from the GUI entirely (and advising people to scale afterwards using other blocks for non-default scaling).

mrjacobagilbert, 2018-03-15:
This behavior has always seemed odd to me as well.

Seems like this behavior could be made more true to name by scaling `d_scale` in the constructor and [here](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/moving_average_XX_impl.cc.t#L95) by d_length.

marcusmueller, 2023-12-21:
I agree, the simple step of setting the default scale to "0", and that translating to scale=1/length would be sensible.