gnuradio/gnuradio #1687: CMake is using deprecated behaviors

dkozel, 2018-03-13:
Currently we enable the old behavior for a set of CMake Policies.

https://github.com/gnuradio/gnuradio/blob/9ae5d4cd3da84ce7e5a0c2abe7fec1ff8d229ae8/CMakeLists.txt#L70-L83

| Policy  | New Behavior Introduced |
| ------ | ------ |
| [26](https://cmake.org/cmake/help/latest/policy/CMP0026.html) | CMake 3.0  |  
| [43](https://cmake.org/cmake/help/latest/policy/CMP0043.html) | CMake 2.8.10  |
| [45](https://cmake.org/cmake/help/latest/policy/CMP0045.html) | CMake 3.0 |
| [46](https://cmake.org/cmake/help/latest/policy/CMP0046.html) | CMake 3.0 |
| [54](https://cmake.org/cmake/help/latest/policy/CMP0054.html) | Cmake 3.1 |

Now, to be more precise about that collumn, the policies were put in at 3.0 and 3.1, the behaviors mostly existed before, so supporting older CMake versions is possible.

# Policy 26

The `GR_ADD_TEST` function seems the big offender.

```
CMake Error at cmake/Modules/GrTest.cmake:45 (get_target_property):
  The LOCATION property may not be read from target "test-gnuradio-pmt".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

Call Stack (most recent call first):
  gnuradio-runtime/lib/pmt/CMakeLists.txt:161 (GR_ADD_TEST)
```

# Policy 43

Only four instances of the old behavior, should be a quick fix.
```
cmake/Modules/GrSetupQt4.cmake
70:SET_PROPERTY(DIRECTORY APPEND PROPERTY COMPILE_DEFINITIONS_DEBUG QT_DEBUG)
71:SET_PROPERTY(DIRECTORY APPEND PROPERTY COMPILE_DEFINITIONS_RELEASE QT_NO_DEBUG)
72:SET_PROPERTY(DIRECTORY APPEND PROPERTY COMPILE_DEFINITIONS_RELWITHDEBINFO QT_NO_DEBUG)
73:SET_PROPERTY(DIRECTORY APPEND PROPERTY COMPILE_DEFINITIONS_MINSIZEREL QT_NO_DEBUG)
```

gateship1, 2020-05-18:
Hello, I am looking to begin contributing to GNURadio. Is this issue still available to work on?
Thank you!

dkozel, 2020-05-26:
@gateship1 Yes! It would be very helpful for you to work on this. The master branch of GNU Radio has a minimum CMake version of 3.8 so you can rely on any features available there.

gateship1, 2020-06-01:
I am using 

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

With both CMake version 3.8 and CMake version 3.13.4, removal of

 # This will suppress developer warnings during the cmake process that can occur 
 # if a newer cmake version than the minimum is used. 
 if(POLICY CMP0026) 
     cmake_policy(SET CMP0026 OLD) 
 endif() 
 if(POLICY CMP0043) 
     cmake_policy(SET CMP0043 OLD) 
 endif() 
 if(POLICY CMP0045) 
     cmake_policy(SET CMP0045 OLD) 
 endif() 
 if(POLICY CMP0046) 
     cmake_policy(SET CMP0046 OLD) 
 endif() 

results a successful build with the errors presented in this issue.  This issue appears resolved and the policies can be removed from the CMakeLists.txt file.

gateship1, 2020-06-09:
https://github.com/gnuradio/gnuradio/pull/3557#partial-pull-merging

closed 2021-03-29 by mormj