gnuradio/gnuradio #1691: new fastnoise implementation fails unpredictably in tests

noc0lour, 2018-03-15:
Building af84fcdca7f4f2123fd0da0a2fb5747e726ff0ea and running test on fastnoise show unpredictable failures in `qa_fastnoise`

```
╭─ /tmp/build-gr  
╰─ ctest --output-on-failure -R fastnoise                       ✔  02:19:06 
Test project /tmp/build-gr
    Start 126: qa_fastnoise
1/1 Test #126: qa_fastnoise .....................   Passed    2.45 sec

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   2.45 sec
╭─ /tmp/build-gr  
╰─ ctest --output-on-failure -R fastnoise                       ✔  02:19:42 
Test project /tmp/build-gr
    Start 126: qa_fastnoise
1/1 Test #126: qa_fastnoise .....................***Failed    2.49 sec
F
======================================================================
FAIL: test_001_test_moments (__main__.test_fastnoise_source)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/andrej/git/gnuradio/gnuradio/gr-analog/python/analog/qa_fastnoise.py", line 85, in test_001_test_moments
    self.assertAlmostEqual(data.var(), 2, places=2)
AssertionError: 1.9934497894857957 != 2 within 2 places

----------------------------------------------------------------------
Ran 1 test in 1.056s

FAILED (failures=1)

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   2.50 sec

The following tests FAILED:
	126 - qa_fastnoise (Failed)
Errors while running CTest
╭─ /tmp/build-gr  
```

noc0lour, 2018-03-15:
I feel checking on correct places is not what we want ....
`2.007` -> correct 
`1.993` -> wrong 

δ is `.007` for both values....

noc0lour, 2018-03-15:
Also I feel this is not the only place where this thing is going wrong. 
What you want to test for is probably the number places for the difference (absolute) to the desired value compared to `0.f`.

marcusmueller, 2018-03-26:
closes by manually merging #1696 .

closed 2018-03-26 by marcusmueller