gnuradio/gnuradio #1694: FFT and IFFT: Hidden scaling

m3x1m0m, 2018-03-15:
It is quite common to define FFT and IFFT as follows: 

```math 
X(a) = \sum_{k=0}^{N-1} x(a) e^(\frac{-j 2\pi k a}{N})
```
```math 
Y(b) = \frac{1}{N} \sum_{k=0}^{N-1} y(b) e^(\frac{j 2\pi k b}{N})
```
where N is the data length.

A nice property of this definition is
```math
X(Y) = x \neq N \cdot x
```
However in GNURadio [the output is scaled by a factor of N](https://gist.github.com/m3x1m0m/9963fe5fd764e44cae1c0a6b4d1b055d). I think it might be an enrichment to have this documented at least.

drmpeg, 2018-03-21:
It's not a GNU Radio thing, it's an FFTW thing.

http://www.fftw.org/faq/section3.html#whyscaled

jdemel, 2018-03-22:
It would be awesome to include this into the `fft block` documentation.

marcusmueller, 2018-03-22:
@jdemel is right, but I don't think "hidden" is really applicable here; I'll change the title and add the "good first issue" tag, hoping to attract people who'll quickly fix the docs.

mbr0wn, 2018-03-23:
This scaling is pretty common and I see no reason to change it. However, I agree we should document it. The block, after all, is not called "FFTW Wrapper Block", so it's up to us to make sure everything is consistent.

m3x1m0m, 2018-03-27:
Thank you everybody for the comments. 

@mbr0wn I can do the documentation. You might assign this issue to me.

m3x1m0m, 2018-04-06:
You are welcome to comment on this documentation change. Thank you.

PR #1749 (2018-04-28): Improved documentation for FFT block.
Fixes #1694
commit 3068ab96 (2018-04-28): Improved documentation for FFT block.

Fixes #1694
closed 2018-04-28 by marcusmueller