gnuradio/gnuradio #1704: GRC Options should allow specifying copyright holder

dkozel, 2018-03-23:
The GRC Options block currently allows setting the Author which results in a comment at the top of the file. I suggest that the header always include the GPL-3.0 SPDX header and a copyright line if a copyright holder is specified.

https://spdx.org/licenses/GPL-3.0-only.html

``` python
#!/usr/bin/env python2
# -*- coding: utf-8 -*-
#
# SPDX-License-Identifier: GPL-3.0
#
##################################################
# GNU Radio Python Flow Graph
# Title: Example Title
# Author: Derek Kozel
# Copyright: 2018 Ettus Research, a National Instruments Company
# Generated: Fri Mar 23 11:16:05 2018
##################################################
```

marcusmueller, 2018-03-23:
I like that it's a sane default and helpful. Can we also include a "generated using GRC {version}" thing?

dkozel, 2018-03-23:
Yes! I had that in my mind this morning but it didn't make it into this issue. That would be very useful information for anyone running the flowgraph three years later.

haakov, 2018-04-25:
I submitted a pull request for this issue above.

How does this look?

```
#!/usr/bin/env python2
# -*- coding: utf-8 -*-
#
# SPDX-License-Identifier: GPL-3.0
#
##################################################
# GNU Radio Python Flow Graph
# Title: Example Title
# Author: Example Author
# Copyright: Example Copyright
# Description: Example Description
#
# Generated: Wed Apr 25 21:29:30 2018
# Generated using GRC version 3.7.12.0
##################################################
```

dkozel, 2018-04-26:
Yes, that output looks good to me. Thanks for the PR!

@marcusmueller There isn't actually a difference between GR version and GRC version right? I want to make sure there's a git hash (when relevant), but looking at the code it should be correct.

haakov, 2018-12-18:
https://github.com/gnuradio/gnuradio/pull/1745 has been merged, I believe this issue can be closed.

mbr0wn, 2018-12-18:
Well then....

closed 2018-12-18 by mbr0wn