gnuradio/gnuradio #1706: Cant build gnuradio get_system_time();

fipso, 2018-03-25:
'''
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.prefix - WARNING - There already is a prefix in `/usr/local'.
Continue using this path Y/[N]? y
Alias `default_prx' already exists, overwrite Y/[N]? y
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.PackageManager - WARNING - A source build for package gnuradio was requested, but binary install was found!
Install gnuradio from source despite binary install available Y/[N]? y
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gnuradio/build
Building:    (100%) [================================================================================================================]
[  4%] Built target volk_obj
[  4%] Built target volk
[  4%] Built target test_all
[  4%] Built target volk-config-info
[  4%] Built target volk_profile
[  4%] Built target pygen_volk_python_volk_modtool_be0ec
[  5%] Built target pygen_volk_python_volk_modtool_b1063
[  5%] Built target pmt_generated
[  6%] Built target gnuradio-pmt
[  9%] Built target gnuradio-runtime
[ 10%] Built target test-gnuradio-runtime
[ 10%] Built target gr_runtime_test
[ 10%] Built target test-gnuradio-pmt
[ 10%] Built target gr_pmt_test
[ 10%] Built target gnuradio-config-info
[ 10%] Built target pmt_swig_swig_doc
[ 10%] Built target _pmt_swig_swig_tag
[ 10%] Built target pmt_swig_gnuradio_runtime_swig_7dd5e
[ 11%] Built target pygen_gnuradio_runtime_swig_32efc
[ 11%] Built target _pmt_swig
[ 11%] Built target runtime_swig_swig_doc
[ 11%] Built target pmt_swig
[ 11%] Built target _runtime_swig_swig_tag
[ 11%] Built target runtime_swig_gnuradio_runtime_swig_7dd5e
[ 11%] Built target _runtime_swig
[ 11%] Built target pygen_gnuradio_runtime_swig_6a954
[ 11%] Built target pygen_gnuradio_runtime_python_gnuradio_04b8e
[ 11%] Built target pygen_gnuradio_runtime_python_gnuradio_gr_f8332
[ 11%] Built target pygen_gnuradio_runtime_python_gnuradio_gru_717d6
[ 11%] Built target pygen_gnuradio_runtime_python_gnuradio_ctrlport_3cc38
[ 11%] Built target pygen_gnuradio_runtime_python_gnuradio_ctrlport_39365
[ 11%] Built target pygen_gnuradio_runtime_python_pmt_28d67
[ 11%] Built target pygen_gnuradio_runtime_examples_mp_sched_13cbd
[ 12%] Built target pygen_gnuradio_runtime_examples_network_753a9
[ 12%] Built target pygen_gnuradio_runtime_examples_volk_benchmark_0202e
[ 13%] Built target blocks_generated_includes
[ 28%] Built target gnuradio-blocks
[ 29%] Built target test-gr-blocks
[ 30%] Built target pygen_gr_blocks_python_blocks_bf940
[ 30%] Built target pygen_gr_blocks_python_grc_gnuradio_88da0
[ 30%] Built target pygen_gr_blocks_python_grc_gnuradio_b4e0e
[ 30%] Built target pygen_gr_blocks_swig_30d22
[ 30%] Built target blocks_swig10_swig_doc
[ 30%] Built target _blocks_swig10_swig_tag
[ 30%] Built target blocks_swig10_gr_blocks_swig_a6e57
[ 30%] Built target pygen_gr_blocks_swig_90afe
[ 30%] Built target blocks_swig9_swig_doc
[ 30%] Built target _blocks_swig9_swig_tag
[ 30%] Built target blocks_swig9_gr_blocks_swig_a6e57
[ 31%] Built target pygen_gr_blocks_swig_f14d2
[ 31%] Built target blocks_swig8_swig_doc
[... 193 lines cut ...]
[ 85%] Built target trellis_swig1_swig_doc
[ 85%] Built target _trellis_swig1_swig_tag
[ 85%] Built target trellis_swig1_gr_trellis_swig_30622
[ 85%] Built target pygen_gr_trellis_swig_f5fac
[ 86%] Built target _trellis_swig1
[ 86%] Built target pygen_gr_trellis_python_trellis_f1421
[ 86%] Built target pygen_gr_trellis_examples_python_6580f
[ 86%] Built target gnuradio-uhd
[ 86%] Built target tags_demo
[ 86%] Built target uhd_swig_swig_doc
[ 86%] Built target _uhd_swig_swig_tag
[ 86%] Built target uhd_swig_gr_uhd_swig_5e3ce
[ 86%] Building CXX object gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In Funktion »PyObject* _wrap_time_spec_t_get_system_time(PyObject*, PyObject*)«:
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:19850:34: Fehler: »get_system_time« ist kein Element von »uhd::time_spec_t«
       result = uhd::time_spec_t::get_system_time();
                                  ^~~~~~~~~~~~~~~
make[2]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/build.make:71: gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:13340: gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all] Fehler 2
make: *** [Makefile:163: all] Fehler 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
'''

dkozel, 2018-03-26:
Hi @fipso. UHD removed the get_system_time function from time_spec_t within the last few months. Errors like this can occur when the python templates don't get correctly rebuilt.

The simplest answer is usually to force a rebuild by deleting the gr-uhd build directory, `/usr/local/src/gnuradio/build/gr-uhd/` in your case. Then rerun the build and it will regenerate the necessary swig files.

As a side note, having both a source and system install of GNU Radio, or any other package, can lead to confusing linking problems. If you've done that intentionally then you're probably already keeping track of what software is built against one or the other.

marcusmueller, 2018-03-31:
I absolutely agree with @dkozel – the problem here is that somehow you're not building with the same headers. As such, it's not a GNU Radio bug, and I'll have to close the issue.

closed 2018-03-31 by marcusmueller

XavierTolza, 2018-07-20:
Deleting the gr-udh folder did not work for me. I have only one PYBOMBS prefix "default" in `~/.pybombs/prefix/default`. After a `rm -R ~/.pybombs/prefix/default/src/gnuradio/build/gr-uhd`, and running `pybombs install gnuradio`, things get re-compiled but I get the same error as previously.

Output from cmd line is the following:
[cmdline_output.txt](https://github.com/gnuradio/gnuradio/files/2213031/cmdline_output.txt)

zbcwilliam, 2018-07-27:
Hi @fipso, i have encountered the same  problem. the uhd(v3.13)has removed uhd::time_spec_t::get_system_time()  and added uhd::get_system_time()；by running " pybombs install gnuradio " command，it calls the uhd::time_spec_t::get_system_time(), thus resulting in the problem. 

I have managed to solve the bug by modifing the "/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx", in which "uhd::time_spec_t::get_system_time()" is called;  here is what i have done: replace " result = uhd::time_spec_t::get_system_time(); "  with  " namespace pt = boost::posix_time;                                                                                                                                                                    pt::ptime time_now = pt::microsec_clock::universal_time();
pt::time_duration time_dur = time_now - pt::from_time_t(0);
result = uhd::time_spec_t(time_t(time_dur.total_seconds()),long(time_dur.fractional_seconds()),double(pt::time_duration::ticks_per_second())); "  then run the command "pybombs install gnuradio", you will do fine.

fipso, 2018-07-29:
Thanks a lot. Will try that

pbybee, 2018-08-20:
@zbcwilliam your replacement worked to get gnuradio installed for me. Thanks

wgaylord, 2018-11-08:
So this seems to still be a problem.  Any solutions to it? Like fixing gnuradio to compile with the version of UHD it compiles from source.

lordcape, 2019-03-08:
@zbcwilliam thanks!

gusosborne, 2019-07-28:
@zbcwilliam thank you sir!!

syvic, 2020-06-23:
Same problem and @zbcwilliam workarround works perfectly. Thanks a lot!

shw1n, 2020-11-24:
I developed this issue after accidentally installing gnuradio from a repository while already having a working installation, and deleting the gr-uhd build folder wasn't working for me. What did work:

`dpkg -l | grep -i uhd` 

This showed me I had an older version of `libuhd-dev` installed, after removing that via 

`dpkg --remove libuhd-dev`

I was able to rebuild gnuradio with no issues