gnuradio/gnuradio #1711: Move const initializations of PMTs into headers

marcusmueller, 2018-03-29:
as #1697 showed, having these in the .cc files can seemingly, under conditions I don't fully grok, lead to hazard. Anyway, having them static const in the header is probably what one wants, anyway. So let's do that.

marcusmueller, 2018-03-30:
ah shoot you're right, and so, so, so wrong! (this discussion revolves around the word "created", which is kinda hard for `pmt_t`, especially for `pmt::symbol`.)

> `static` outside of class members leads to those symbols being local to the compilation unit (read: .cc file) which makes them unavaiable in other compilation units.

No, that's neither true (as you can see by the fact that these `gr::uhd` members were *declared*), nor the point.

Distinguish declaration, definition, allocation, initialization. (and "existence", but that term is a C++ fuckup of its own)

> Putting them in the header will define them locally in every other unit where the header is included.

The *symbol* (in the C calling convention sense of "sym") is known by declaring it in the header. 

However, **the first call** to `pmt::symbol("some string")` will insert `some string`'s PMT representation into a hash table, which isn't even value-ordered, but hybridly value-chronologically ordered. You get a kinda-smart pointer to that PMT back. (Remember that `pmt_t`, as much as that sounds like a true C type, is a `boost::intrusive_ptr<pmt_base>`).

So, the value of `pmt::symbol("andrej")` depends on who called it first, and where that first call generated the PMT object. (especially, it's not a `constexpr` of any kind, and can't be determined at compile time at all. You actually have to call the function at run time to get that object.)

So, semantically, two different program runs will give you two very different `pmt::symbol("andrej")` simply by fortune of that being kind of a unique ptr, and also, because the hash map containing the object being pointed to will be different. So, what we want is for these to be generated once, and only once, if possible, with the first instantiation of a class that contains a static data member.

But, hm, C++ [won't play that game](http://en.cppreference.com/w/cpp/language/storage_duration)! 

> `static` storage duration. The storage for the object is allocated when the program begins and deallocated when the program ends. Only one instance of the object exists. All objects declared at namespace scope (including global namespace) have this storage duration, plus those declared with static or exter

So, ok, we're allocating memory for each `static pmt_t` at class (or namespace) level when the program starts. Not really cool, but how it is, then. It's just a couple of bytes, right?

The question now is, **when** exactly is `pmt::symbol` called then?

Thus, he quoteth [ISO/IEC 14882:1998]() (c++98), from the book of classes, chapter on static members, section on data members, stanza on lifetime (9.4.2 §7):

> Static data members are initialized and destroyed exactly like non-local objects (3.6.2, 3.6.3).

{very loud explicative, actually shouted whilst opening section 3.6.2, and realizing what I've done} I introduced the *Static Initialization Order Fiasco* (SIOF, google it) into GNU Radio.

Long story short: Hashtable and other PMT things are static.

`CMD_GAIN_KEY` is static.

Initialization order of `static` members in different compilation orders is undefined. And will, in fact, be only defined at run time, if these not happen to be in the same shared object.

You see where this is going. It's absolutely random whether the PMT hashtable already exists at the point these things are actually initialized. If it's not there: happy segfault day!

And that's exactly what *actually* caused #1688. By putting the definitions in the header, we just got lucky at runtime, as far as I can tell.

Conclusion:

We can't have `static` PMT symbols. This of course has very dark consequences for PMT as ... polymorphic type library, but I can't change that now.

marcusmueller, 2018-03-30:
… which means I have to revert a lot of the (I thought was good) changes I've done in #1628. Arrgh!

noc0lour, 2018-03-30:
So, my observation was from the top of my head and not correct, but luckily it helped you think about the problem :+1:

PR #1713 (2018-03-30): Fix the potential for Static Initialization Order Fiasco
Found thanks to #1697.

Closes #1711.
closed 2018-03-30 by marcusmueller