gnuradio/gnuradio #1720: Compile error when using pybombs

mugpahug, 2018-04-05:
When performing a clean install  of gnu radio in a prefix using
 pybombs prefix init -a default prefix/default -R gnuradio-default

The following compile error appears
[grc-bug.txt](https://github.com/gnuradio/gnuradio/files/1878640/grc-bug.txt)

chie4hao, 2018-04-08:
same issue, see this
[https://github.com/EttusResearch/uhd/commit/2c9087d120f0cb5ede1436cd0508cf92c3bcd68e#diff-b57ef0f419076115f5b6c54f5fc92691](https://github.com/EttusResearch/uhd/commit/2c9087d120f0cb5ede1436cd0508cf92c3bcd68e#diff-b57ef0f419076115f5b6c54f5fc92691)

noc0lour, 2018-04-08:
The mentioned UHD fix is correct and prevents libUHD headers leaking C++11 code to non-C++11 headers. Which is ultimately not a bug in GNU Radio.

closed 2018-04-08 by noc0lour