gnuradio/gnuradio #1735: Sending last message WHILE a flowgraph is closing.

reiinakano, 2018-04-17:
While a flow graph is running, it's easy to have an external program get data from various blocks via messages through the ZMQ message sink. My question is what if the data is generated AFTER a flow graph is closed e.g. a png image of the full waterfall plot?

When I modified the `stop` function of a block to pass directly to the ZMQ message sink, it seems that at that point, the ZMQ message sink has closed since it doesn't process/send the message anymore. Is there a way to keep a block from closing until all other blocks that send messages to it have closed? Is the `stop` function the correct place to put this logic?

willcode, 2018-05-05:
No, stop() is not coordinated. Everything just stops via thread interrupts. Most blocks don't do anything on stop().

willcode, 2020-10-28:
This may be addressed by https://github.com/gnuradio/gnuradio/pull/3866, and is similar to https://github.com/gnuradio/gnuradio/issues/3872. @reiinakano can this be closed?

marcusmueller, 2023-12-21:
this is a support request, not a bug report, I'm afraid, and we're currently cleaning up the issue tracker

closed 2023-12-21 by marcusmueller