gnuradio/gnuradio #1736: Compiler error: too few arguments to function volk_32f_8u_polarbutterfly_32f(llrs, u, block_power(), stage, u_num, row)

minecraft2048, 2018-04-18:
I'm trying to install gnuradio from source using Arch Linux pkgbuild system, and it errors with this compiler error:

```
[ 52%] Generating random_uniform_source_b_impl.cc, random_uniform_source_s_impl.cc, random_uniform_source_i_impl.cc
[ 52%] Generating noise_source_s_impl.h, noise_source_i_impl.h, noise_source_f_impl.h, noise_source_c_impl.h
[ 52%] Generating noise_source_s_impl.cc, noise_source_i_impl.cc, noise_source_f_impl.cc, noise_source_c_impl.cc
[ 52%] Generating fastnoise_source_s_impl.h, fastnoise_source_i_impl.h, fastnoise_source_f_impl.h, fastnoise_source_c_impl.h
[ 52%] Generating fastnoise_source_s_impl.cc, fastnoise_source_i_impl.cc, fastnoise_source_f_impl.cc, fastnoise_source_c_impl.cc
[ 52%] Generating sig_source_s_impl.h, sig_source_i_impl.h, sig_source_f_impl.h, sig_source_c_impl.h
[ 52%] Generating sig_source_s_impl.cc, sig_source_i_impl.cc, sig_source_f_impl.cc, sig_source_c_impl.cc
[ 53%] Generating random_uniform_source_b_impl.h, random_uniform_source_s_impl.h, random_uniform_source_i_impl.h
Scanning dependencies of target gnuradio-analog
/tmp/yaourt-tmp-feanor/aur-gnuradio-git/src/gnuradio/gr-fec/lib/polar_decoder_common.cc: In member function ‘void gr::fec::code::polar_decoder_common::butterfly_volk(float*, unsigned char*, int, int, int)’:
/tmp/yaourt-tmp-feanor/aur-gnuradio-git/src/gnuradio/gr-fec/lib/polar_decoder_common.cc:128:81: error: too few arguments to function
         volk_32f_8u_polarbutterfly_32f(llrs, u, block_power(), stage, u_num, row);
                                                                                 ^
make[2]: *** [gr-fec/lib/CMakeFiles/gnuradio-fec.dir/build.make:1071: gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_common.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 53%] Building CXX object gr-analog/lib/CMakeFiles/gnuradio-analog.dir/noise_source_s_impl.cc.o
[ 53%] Building CXX object gr-analog/lib/CMakeFiles/gnuradio-analog.dir/noise_source_i_impl.cc.o
[ 53%] Building CXX object gr-analog/lib/CMakeFiles/gnuradio-analog.dir/noise_source_f_impl.cc.o
```
Currently the git head commit is at commit d19965e42d12bba66e39fd20794c69c69c48b31f.

jdemel, 2018-04-18:
Update VOLK to 1.4, GR 3.7.12 requires this version.

minecraft2048, 2018-04-18:
Confirmed, using libvolk-git from AUR instead of official Arch Linux repo fixes this problem, thanks

closed 2018-04-18 by minecraft2048