gnuradio/gnuradio #1742: Error importing GNU Radio's wxgui. Please make sure gr-wxgui is installed.

cn0xroot, 2018-04-20:
Hi，GNURadio project.
I has installed gnuradio-devel via macport on mac,but when i use osmocomsdr

it doesn't work:Error importing GNU Radio's wxgui. Please make sure gr-wxgui is installed.

```
gnuradio-config-info --enabled-components
python-support;testing-support;doxygen;sphinx;gnuradio-runtime;gr-blocks;gnuradio-companion;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-atsc;gr-audio;* jack;* osx;* portaudio;gr-channels;gr-noaa;gr-pager;gr-qtgui;gr-trellis;gr-uhd;gr-utils;gr-video-sdl;gr-vocoder;gr-fcd;gr-wavelet;gr-zeromq
```
Is there any people can help me to fix it?
what should i do? thx!

cn0xroot, 2018-04-20:
Hi @michaelld it seem like gr-osmosdr has something error

```
port info gr-osmosdr
gr-osmosdr @20170611_6 (science, comms)
Variants:             [+]airspy, [+]bladeRF, debug, [+]docs, [+]fcdproplus, [+]hackrf, [+]rtlsdr, [+]sdrplay, [+]soapysdr, [+]swig, [+]uhd,
                      universal

Description:          Includes OSMO SDR support GNU Radio source and sink blocks in C++, Python, and GNU Radio Companion (grc). This port
                      also offers a wrapper functionality for the FunCube Dongle, FunCube Dongle Pro+, HackRF, RFSpace, Ettus UHD, and
                      rtl-sdr radios and software. By using gr-osmosdr source you can take advantage of a common software API in your
                      applications independent of the underlying radio hardware. This port is kept up with the gr-osmosdr GIT 'master'
                      branch, which is typically updated weekly to monthly, providing compatibility with GNU Radio release 3.7 API: the
                      gnuradio and gnuradio-devel ports.
Homepage:             http://sdr.osmocom.org/trac/wiki/GrOsmoSDR

Build Dependencies:   cmake, pkgconfig, doxygen, graphviz
Library Dependencies: gnuradio, boost, volk, python27, uhd, swig-python, hackrf, gr-fcdproplus, rtl-sdr, bladeRF, airspy, SoapySDR, sdrplay
Test Dependencies:    cppunit
Platforms:            darwin, macosx
License:              GPL-3
Maintainers:          Email: michaelld@macports.org, GitHub: michaelld
                      Policy: openmaintainer
```

michaelld, 2018-04-20:
what does `port installed gnuradio-devel` return?

michaelld, 2018-04-20:
please note that gr-wxgui is going away with the 3.8 release, and there are enough issues with it (and gr-qtgui works so much better & is roughly a drop-in replacement) that I disabled auto-install of gr-wxgui with the gnuradio-devel port. You can still try to install it via `sudo port install gnuradio-devel +wxgui`.

cn0xroot, 2018-04-20:
```
port installed gnuradio-devel
The following ports are currently installed:
  gnuradio-devel @20180405_1+docs+grc+jack+log4cpp+logging+portaudio+qtgui+sdl+swig+uhd+wavelet+zeromq (active)
```

michaelld, 2018-04-20:
yeah so you have the default gnuradio-devel install. try installing it with +wxgui.

BTW> can you please attach the commandline & resulting output from whatever you did that resulted in the gr-wxgui required error?

cn0xroot, 2018-04-20:
when i use osmcom_fft,i get the error
```
osmocom_fft -f 315e6 -s 2e6
Error importing GNU Radio's wxgui. Please make sure gr-wxgui is installed.
osmocom_fft --help
Error importing GNU Radio's wxgui. Please make sure gr-wxgui is installed.
```
and i have some grc file , it base wxgui.

michaelld, 2018-04-20:
Ah; OK that make sense. Hmmm .. yeah I'll need to fix issue that somewhere in gr-osmosdr or gnuradio [-devel] ... will add it to my queue & think deep thoughts ...

michaelld, 2018-04-20:
I actually have `gnuradio-devel +wxgui` installed and it works; so does `gr-osmosdr` then & `osmosdr_fft` too. Go figure.

cn0xroot, 2018-04-20:
Thanks for your quickly replay,i think it will work fine when i have` gnuradio-devel +wxgui`.
GNURaddio is a very nice project! 
Have a nice day.

michaelld, 2018-04-20:
Glad to help. Please report back here your success in getting `gnuradio-devel +wxgui` to install. You might have to rebuild gr-osmosdr with the updated GR; might not too. Try your `osmosdr_fft` & see if it works after the GR update; it very well might in which case you're good to go.

cn0xroot, 2018-04-20:
oh,i meet another error：

```
osmocom_fft
Traceback (most recent call last):
  File "/opt/local/bin/osmocom_fft", line 48, in <module>
    from gnuradio.wxgui import forms
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/wxgui/forms/__init__.py", line 36, in <module>
    from forms import \
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/wxgui/forms/forms.py", line 64, in <module>
    class _form_base(pubsub, wx.BoxSizer):
TypeError: Error when calling the metaclass bases
    multiple bases have instance lay-out conflict
```

mbr0wn, 2018-04-28:
This kind of discussion is better suited for the mailing list, it is not an issue with the codebase as far as I can tell. Closing now.

closed 2018-04-28 by mbr0wn

tresacton, 2019-07-04:
I hate necro-posting, but thought I would anyway in case it's of use to the next person that comes along... 

This issue came about for me when I went to install gr-osmosdr via macports, without having first installed gnu radio (I figured the port for gr-osmosdr would handle the dependencies correctly - which it didn't seem to for me on Mac OSX Mojave).

The installation for gr-osmosdr via macports seems to depend on the user having already run:

`sudo port install gnuradio +grc +swig +wxgui +qtgui
`
I did not need to re-install gr-osmosdr after installing gnuradio

loewal, 2020-03-20:
Hopefully not off-topic:
I tried to install gr-osmosdr through Macports (clean!).
Next error, which I can't ...understand:

:info:build #pragma message("The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead")
:info:build         ^
:info:build /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gr-osmosdr/gr-osmosdr/work/gr-osmosdr-4d83c6067f059b0c5015c3f59f8117bbd361e877/lib/rtl/rtl_source_c.cc:224:9: error: use of undeclared identifier 'rtlsdr_set_bias_tee'
:info:build   ret = rtlsdr_set_bias_tee(_dev, bias_tee);
:info:build         ^
:info:build /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gr-osmosdr/gr-osmosdr/work/gr-osmosdr-4d83c6067f059b0c5015c3f59f8117bbd361e877/lib/rtl/rtl_source_c.cc:482:26: error: use of undeclared identifier 'RTLSDR_TUNER_R828D'; did you mean 'RTLSDR_TUNER_R820T'?
:info:build     } else if ( tuner == RTLSDR_TUNER_R828D ) {
:info:build                          ^~~~~~~~~~~~~~~~~~
:info:build                          RTLSDR_TUNER_R820T
:info:build /usr/local/include/rtl-sdr.h:178:2: note: 'RTLSDR_TUNER_R820T' declared here
:info:build         RTLSDR_TUNER_R820T
:info:build         ^
:info:build 1 warning and 2 errors generated.
:info:build make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o] Error 1
:info:build make[2]: *** Waiting for unfinished jobs....
:info:build 1 warning generated.
:info:build 1 warning generated.
:info:build make[2]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gr-osmosdr/gr-osmosdr/work/build'
:info:build make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
:info:build make[1]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gr-osmosdr/gr-osmosdr/work/build'
:info:build make: *** [all] Error 2
:info:build make: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gr-osmosdr/gr-osmosdr/work/build'
:info:build Command failed:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gr-osmosdr/gr-osmosdr/work/build" && /usr/bin/make -j4 -w all VERBOSE=ON
:info:build Exit code: 2
:error:build Failed to build gr-osmosdr: command execution failed
:debug:build Error code: CHILDSTATUS 33400 2
:debug:build Backtrace: command execution failed
:debug:build     while executing
:debug:build "system {*}$notty {*}$nice $fullcmdstring"
:debug:build     invoked from within
:debug:build "command_exec build"
:debug:build     (procedure "portbuild::build_main" line 8)
:debug:build     invoked from within
:debug:build "$procedure $targetname"

Why on earth ??