gnuradio/gnuradio #1746: No Support for Loading Filter Taps From Files

m3x1m0m, 2018-04-26:
As far as I know there is no support for loading filter taps from files. For this reason, I wrote a small patch for this missing functionality: https://github.com/m3x1m0m/gnuradio/tree/file_taps_loader

However, I am not sure how I can get it into the "File Operators" branch. How is this done? 

Thank you for your help.

willcode, 2018-04-26:
You can use Python expressions for parameters in GRC. For example, in 
the FFT Filter block, "Taps" could be:

   numpy.fromfile('taps.dat', dtype=numpy.complex64)

Make sure you add an Import block containing "import numpy". Create the 
data in a Python shell using something like:

   import numpy
   a=numpy.array((1+1j), dtype=numpy.complex64)
   with open('taps.dat','w') as f: a.tofile(f)

No extra blocks required.

m3x1m0m, 2018-04-27:
@willcode Thank you very much for your comment. 

I might be totally wrong, but numpy, as I experienced, is not capable of correctly interpreting files produced by another part of gnuradio, the filter tool. Also, from my point of view, this is another hack, one has to be aware of. (As a new user this can be frustrating.)  

This is an example flowgraph trying to use numpy and a file generated with the filter tool:
https://gist.github.com/m3x1m0m/ccebd874829b9f4ec08b06b45e9fbc70

The taps:
https://gist.github.com/m3x1m0m/a2e3bb432f503c53920a27aebdc77142

I could not find a solution, working out of grc, to produce the correct data format, that the interp_fir_filter_ccc_make is looking for (apparently std::vector<gr_complex>). Also, why should everybody has to go through this?  

Instead, I propose to introduce a possibility to integrate the designed filters, that is obvious and not possible to overlook, hence a small, simple block. To summarize, I think using gnuradio, should be fixing less while working more on what you actually want to achieve, (which is not finding workarounds to import filter taps).

m3x1m0m, 2018-04-30:
Thank you again @willcode for your input. I moved the block to gr-filter and edited the one example where these other convenience blocks are used. I have no clue how to open the filter design tool directly from grc. There is no template mechanism available in grc  so far for this kind of behavior, is there? I can look further into this.

m3x1m0m, 2018-05-21:
@marcusmueller Is there something else I should do, that this can be merged into the master branch?

devnulling, 2020-01-22:
Fixed with: https://github.com/gnuradio/gnuradio/commit/f181d2899dc9181963d591e33856afa6aa8c90ae#diff-d6b86fb2420014db580944b8af40d493

closed 2020-01-22 by devnulling

canopus19652000, 2020-09-17:
Workaround does exist for version 3.7.13.5 at least (I am using this version on Windows)
One could write Re and Im parts of complex filter in to two different "real" files. Then, one could read both files in to FFT block and combine them in to complex taps.

For example:

numpy.fromfile('Re_file', sep=',') + numpy.fromfile(‘Im_file', sep=',')*1.0j 

I use next python method to create these files (Irrelevant python statement are omitted):

coeff = filter.firdes.complex_band_pass_2( Gain, SampRate, LCO, HCO, TransWidth, Atten, Window, Beta )
res =  numpy.asarray(coeff)
Re_file.write(str(tuple(res.real)).strip(')('))
Im_file.write(str(tuple(res.imag)).strip(')('))

No needs to wait release 3.9