gnuradio/gnuradio #1751: Gnuradio Companion 3.7.11-10 Crashing

anthonynacz, 2018-04-30:
I have lately upgraded my ubuntu from 16.04 to 18.04 , to be able to use newer release of gnuradio . The latest being 3.7.11-10 . The problem with it is that it keeps crashing ,immediately  after running that I cannot get any use out of it . I appreciate any help

marcusmueller, 2018-05-01:
Since it works out of the box on all other Ubuntu installations, this is most likely something wrong with your specific machine. Maybe you have something compiled yourself that now links against the wrong version of GNU Radio?

Anyway, we'd really have to sit in front of your PC to diagnose that - and since it's sadly not an issue with GR itself but with your setup, I'm regretting to inform you that I'll close this issue.

If this happens on a clean slate installation, please do reopen.

closed 2018-05-01 by marcusmueller

drmpeg, 2018-05-02:
A build from the latest source works fine here on 18.04. However, when switching to the old Unity desktop, GNU Radio Companion crashes as soon as you mouse over it.