gnuradio/gnuradio #1755: Test C++14, 20 compatibility

marcusmueller, 2018-05-01:
https://stackoverflow.com/questions/50112655/why-has-assignment-to-a-gr-complex-variable-gnu-radio-stopped-working-for-me

I guess that C++14 or 20 changes std::complex semantics, and we'll have fun.

noc0lour, 2018-05-01:
And really removes other keywords which are deprecated in C++11 :)

ncorgan, 2020-04-19:
The non-const-ref real() and imag() functions were non-standard libstdc++ extensions, eventually removed (even for C++11). When Price references upgrading his stack, I assume there was a compiler upgrade as well.

See the commit removing them here: https://gcc.gnu.org/git/?p=gcc.git&a=commit;h=23ed71c60c5a2783c092a00cd39133c2f057d4dc
See the bickering afterward here: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=51013

ncorgan, 2020-04-19:
Just from quick checks:

GNU Radio compiles and tests fine with C++14, but it will need to enforce a minimum version of Log4Cpp for C++17.

marcusmueller, 2020-11-12:
all results (incl Win/anaconda) are positive!

closed 2020-11-12 by marcusmueller