gnuradio/gnuradio #1760: Build failed

yangdinglou, 2018-05-04:
/home/ydl/gnuradio/gr-fec/lib/polar_decoder_common.cc: In member function ‘void gr::fec::code::polar_decoder_common::butterfly_volk(float*, unsigned char*, int, int, int)’:
/home/ydl/gnuradio/gr-fec/lib/polar_decoder_common.cc:128:95: error: too many arguments to function
 8u_polarbutterfly_32f(llrs, u, block_size(), block_power(), stage, u_num, row);
                                                                              ^
gr-fec/lib/CMakeFiles/gnuradio-fec.dir/build.make:1070: recipe for target 'gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_common.cc.o' failed
**Environment: gcc 5.5 ubuntu16.04**

I used to build successfully last year without modify.

dkozel, 2018-05-04:
@yangdinglou Have you updated Volk as well as GNU Radio? There was a recent API change. If you are using git you will need to update the submodule as well as the main repo.
`git submodule update`

If that does not solve your issue could you please send a message to the mailing list?
https://wiki.gnuradio.org/index.php/ReportingErrors

Thanks!

closed 2018-05-04 by dkozel

yangdinglou, 2018-05-16:
@dkozel Thankyou! And It's my fault to not reply to you. I noticed the API update and modified the script that day. It was my first time to pull issue, so I don't know I could comment on closed issue.
Sorry for replying to late.