gnuradio/gnuradio #1762: Socket PDU (as server) runaway memory consumption

regulatre, 2018-05-05:
I believe the socket PDU has a memory leak that is triggered by a client which disconnects abruptly. 

Here's the scenario: 
I have two flowgraphs, one is acting as a server and one as the client. Normally they connect and transfer data and all is well.. 

But recently, I've been having problems with the client (unrelated to the server side bug), and it tends to crash and then re-connect to the server flowgraph about once every minute or two. This leads to the bug I'm reporting here. I believe the crashing of the client flowgraph bypasses the normal orderly shutdown of the socket, and perhaps leaves it in a state where the client is dead and the server is still trying to send data (that's just a theory). 

The server socket PDU block gets into this funky state after a few dozen of these abrupt disconnects of the TCP connection. I track the RSS size of the server flow graph app and I can observe a steady increase in memory consumption that goes up by about 5 MB per minute until the system runs out of resources and then a range of symptoms are exhibited by the OS - OOM killer goes into effect, and in at least one case "IOError: [Errno 24] Too many open files"  The source flow graph runs at a sample rate of 48KHz and I suspect data from the source is just getting buffered somewhere until the system runs out of resources to store any more data. 

I've observed this behavior on both Raspberry Pi 3 (ARM6) architecture, as well as a native x86_64 Linux machine. The Pi scenario traverses two hops in a LAN, and the x86_64 scenario has both flowgraphs on the same host. 

Sorry I'm not handy enough with C++ to help find the bug, but my gut tells me there's something up with the socket PDU block (server mode), and has to do with not detecting the abnormal shutdown of a client connection.

willcode, 2018-05-05:
It looks like the problem is in the use of async io, where tcp_connection.send() is called but the results of async_write() are not tracked.

marcusmueller, 2023-12-21:
Socket PDU might have survived from 3.7 days and was always relatively convoluted – needs checking