gnuradio/gnuradio #1764: vector sink "malloc(): memory corruption" / object was probably modified after being freed

regulatre, 2018-05-05:
(originally posted to gnuradio-discuss)
 
The Vector Sink is coming in very handy for some experimentation I'm doing. I'm analyzing the output of an FFT block which terminates into a float vector sink. every few seconds from a thread which then calls reset() to clear the contents in preparation for another read. This seems problematic as the program crashes quite frequently after seconds to minutes of operation. Based on my layman's perspective it seems to be a memory leak in the vector block.

I tried many things over the last couple of days. Nothing seems to mitigate it. For example self.lock and self.unlock to lock the flowgraph before reading from the vector (results in flowgraph never starting back up, seems to be a whole different issue)... I tried using python copy.deepcopy to make a copy of the vector contents before using it but that didn't help either.

When the exception occurs, it seems to happen right after resetting the vector sink.

My flowgraph doesn't run standalone and requires a number of other applications to function but I'll work on getting it up to github for review if that helps

Gnuradio 3.7.11.1 on a CentOS VM (Linux 3.10.0-514.el7.x86_64 #1 SMP Tue Nov 22 16:42:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux)

BTW when the exception occurs on OSX the error is:

`Python(86100,0x70000eab9000) malloc: *** error for object 0x7f94bf9042b8: incorrect checksum for freed object - object was probably modified after being freed.`

Stack Trace samples from Linux (they are all quite similar to eachother with only slight differences in the  addresses):

```
*** Error in `python': malloc(): memory corruption: 0x00007fa0cc019ba0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7dd4d)[0x7fa14f448d4d]
/lib64/libc.so.6(__libc_malloc+0x4c)[0x7fa14f44afbc]
/lib64/libstdc++.so.6(_Znwm+0x1d)[0x7fa145c0e0cd]
/usr/local/lib64/python2.7/site-packages/gnuradio/blocks/_blocks_swig1.so(_ZNSt6vectorIfSaIfEEaSERKS1_+0xd4)[0x7fa13df58e64]
/usr/local/lib64/python2.7/site-packages/gnuradio/blocks/_blocks_swig1.so(+0x12c8a0)[0x7fa13deea8a0]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6df0)[0x7fa150195af0]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7fa1501954bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7fa1501954bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7ed)[0x7fa150197e3d]
/lib64/libpython2.7.so.1.0(+0x7088d)[0x7fa15012188d]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7fa1500fc8e3]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x17fd)[0x7fa1501904fd]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7fa1501954bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7fa1501954bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7ed)[0x7fa150197e3d]
/lib64/libpython2.7.so.1.0(+0x70798)[0x7fa150121798]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7fa1500fc8e3]
/lib64/libpython2.7.so.1.0(+0x5a8d5)[0x7fa15010b8d5]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7fa1500fc8e3]
/lib64/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47)[0x7fa15018e6f7]
/lib64/libpython2.7.so.1.0(+0x1155c2)[0x7fa1501c65c2]
/lib64/libpthread.so.0(+0x7dc5)[0x7fa14fe9cdc5]
/lib64/libc.so.6(clone+0x6d)[0x7fa14f4c276d]
======= Memory map: ========
00400000-00401000 r-xp 00000000 fd:00 242126                             /usr/bin/python2.7
00600000-00601000 r--p 00000000 fd:00 242126                             /usr/bin/python2.7
--
*** Error in `python': malloc(): memory corruption: 0x00007f03ac00f640 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7dd4d)[0x7f04382b6d4d]
/lib64/libc.so.6(__libc_malloc+0x4c)[0x7f04382b8fbc]
/lib64/libstdc++.so.6(_Znwm+0x1d)[0x7f042ea7c0cd]
/usr/local/lib64/python2.7/site-packages/gnuradio/blocks/_blocks_swig1.so(_ZNSt6vectorIfSaIfEEaSERKS1_+0xd4)[0x7f0426dc6e64]
/usr/local/lib64/python2.7/site-packages/gnuradio/blocks/_blocks_swig1.so(+0x12c8a0)[0x7f0426d588a0]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6df0)[0x7f0439003af0]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7f04390034bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7f04390034bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7ed)[0x7f0439005e3d]
/lib64/libpython2.7.so.1.0(+0x7088d)[0x7f0438f8f88d]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f0438f6a8e3]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x17fd)[0x7f0438ffe4fd]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7f04390034bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x67bd)[0x7f04390034bd]
/lib64/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7ed)[0x7f0439005e3d]
/lib64/libpython2.7.so.1.0(+0x70798)[0x7f0438f8f798]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f0438f6a8e3]
/lib64/libpython2.7.so.1.0(+0x5a8d5)[0x7f0438f798d5]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f0438f6a8e3]
/lib64/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47)[0x7f0438ffc6f7]
/lib64/libpython2.7.so.1.0(+0x1155c2)[0x7f04390345c2]
/lib64/libpthread.so.0(+0x7dc5)[0x7f0438d0adc5]
/lib64/libc.so.6(clone+0x6d)[0x7f043833076d]
======= Memory map: ========
00400000-00401000 r-xp 00000000 fd:00 242126                             /usr/bin/python2.7
00600000-00601000 r--p 00000000 fd:00 242126                             /usr/bin/python2.7
```

willcode, 2018-05-05:
Not sure whether this is in vector_sink, but there's potential bug there. In data() and tags(), a scoped_lock is used, but only it only protects the d_data and d_tags vectors, not the contents of those vectors. Just as a guess, there is a reader iterating over the vector at the same time reset() is called. Try commenting out any vector reads (leave the resets in) to see if this is the case.

regulatre, 2018-05-05:
ok! Will try that now and let you know how it goes

regulatre, 2018-05-05:
I disabled the read and left the reset as instructed. Error still present

Tested on x86_64 machine, 3.7.11 

First try: 
Result: after 90 seconds (about 8 resets): `Python(50515,0x70000f07b000) malloc: *** error for object 0x7fb4ef20cac8: incorrect checksum for freed object - object was probably modified after being freed.`

Second try: 
Result: after 146 seconds (about 12 resets): `Python(65898,0x70000178b000) malloc: *** error for object 0x7fb7df8b4838: incorrect checksum for freed object - object was probably modified after being freed.`

willcode, 2018-05-05:
A sample application would help (the minimum needed to reproduce the 
problem).

regulatre, 2018-05-06:
Thank you @willcode for taking the time to help me troubleshoot this. After trying a lot of things I came up with a simple modification to your app that leads to the same issue. I added an FFT in the path. 

Updated vector test python flowgraph: https://gist.github.com/regulatre/c307abe5e76024cd715c7aabf5185e1b

```
Connections...
Connections done.
Press Enter to quit: 0 8192 0
1 16352 0
2 8160 0
3 8160 0
4 8192 0
5 16352 0
6 8160 0
7 8192 0
8 16352 0
9 8160 0
10 8192 0
11 8192 0
12 8160 0
Python(74633,0x70000df1c000) malloc: *** error for object 0x7fdc83009e00: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
13Abort trap: 6
```

again
```
157 8192 0
Python(75015,0x70000047c000) malloc: *** error for object 0x7f812e0d0e00: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
158Abort trap: 6
```

again
```
153 8160 0
Python(75030,0x70000b3ce000) malloc: *** error for object 0x7fa2de3aea00: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
154Abort trap: 6
```

again
```
853 8192 0
Python(75046,0x700007b66000) malloc: *** error for object 0x7fa5aa3c3800: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
854Abort trap: 6
```

again
```
59 8160 0
Python(75073,0x700005a55000) malloc: *** error for object 0x7faa63114600: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
60Abort trap: 6
```

willcode, 2018-05-06:
It turns out that the commit that added locking was added after 3.7.11, so you're seeing a race between work and reset. Your version of the script worked fine on the latest master code, so I think you'll have to upgrade. This diff only affects 2 files, so you could apply it to your version if needed.

commit b0699f7dc6b7e6ed278749ee03245225db8d2a7a
Author: Marcus Müller <marcus.mueller@ettus.com>
Date:   Sun Aug 27 18:14:53 2017 +0200

    added a mutex to the data extraction, clear and work methods

regulatre, 2018-05-06:
recompiling gnuradio now to confirm.

regulatre, 2018-05-06:
I ran into compile errors on maint branch so switched to master and that compiled fine but it's lacking the above commit. I'll see if I can cherry pick that commit to master locally

regulatre, 2018-05-07:
I'm stuck trying to recompile on the master branch - I could use some hints - I'm stuck with what looks like a volk vs gnuradio core code/api mismatch. 

I'm on the volk origin/master branch 
```
volk$ git describe 
v1.4-9-g297fefd

* 297fefd -(10 days ago) Fix 32u_reverse_32u for ARM. . Philip Balister (HEAD, origin/master, origin/HEAD)

```

And gnuradio is up to date. I'm on the master branch:
```
gnuradio$ git describe
3.7.12git-430-g82d0a6b

* 82d0a6b -(4 weeks ago) gr-newmod: Pylint fixes in python scripts . Swapnil Negi (HEAD, origin/master, origin/HEAD, master)
```

The error: 
```[ 17%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/test_tag_variable_rate_ff_impl.cc.o
/home/brad/gr/gnuradio/gr-blocks/lib/test_tag_variable_rate_ff_impl.cc: In member function ‘virtual int gr::blocks::test_tag_variable_rate_ff_impl::general_work(int, gr_vector_int&, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/brad/gr/gnuradio/gr-blocks/lib/test_tag_variable_rate_ff_impl.cc:83:48: error: ‘UINT64_MAX’ was not declared in this scope
         if (xoroshiro128p_next(d_rng_state) > (UINT64_MAX / 2)){
                                                ^
/home/brad/gr/gnuradio/gr-blocks/lib/test_tag_variable_rate_ff_impl.cc:100:50: error: ‘UINT64_MAX’ was not declared in this scope
           if (xoroshiro128p_next(d_rng_state) > (UINT64_MAX / 2)){
                                                  ^
make[2]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/test_tag_variable_rate_ff_impl.cc.o] Error 1
make[1]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/all] Error 2
make: *** [all] Error 2
```

regulatre, 2018-05-07:
I'm going to blow away everything and start over using the build-gnuradio script with -m (for clone master branch) option.

regulatre, 2018-05-07:
Same compile error. Any suggestions?

```
[ 60%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/transcendental_impl.cc.o
/home/brad/gr/gnuradio/gr-blocks/lib/test_tag_variable_rate_ff_impl.cc: In member function ‘virtual int gr::blocks::test_tag_variable_rate_ff_impl::general_work(int, gr_vector_int&, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/brad/gr/gnuradio/gr-blocks/lib/test_tag_variable_rate_ff_impl.cc:83:48: error: ‘UINT64_MAX’ was not declared in this scope
         if (xoroshiro128p_next(d_rng_state) > (UINT64_MAX / 2)){
                                                ^
/home/brad/gr/gnuradio/gr-blocks/lib/test_tag_variable_rate_ff_impl.cc:100:50: error: ‘UINT64_MAX’ was not declared in this scope
           if (xoroshiro128p_next(d_rng_state) > (UINT64_MAX / 2)){
                                                  ^
make[2]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/test_tag_variable_rate_ff_impl.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/all] Error 2
make: *** [all] Error 2
```

noc0lour, 2018-05-07:
hm, looks like a missing `#include <stdlib.h>` / `#include <cstdlib>` what system/compiler are you using?

regulatre, 2018-05-08:
@noc0lour  Thanks for the hint, I'll see if I can find which file is missing that include statement. Meanwhile here is the system/compiler info:

CentOS 7.3

```
$ uname -a
Linux 3.10.0-514.el7.x86_64 #1 SMP Tue Nov 22 16:42:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
[brad@sdr gnuradio]$ gcc --version
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-16)
Copyright (C) 2015 Free Software Foundation, Inc.
```

regulatre, 2018-05-08:
Good news - I did a yum update, bringing my platform up to CentOS 7.4 and kernel 3.10.0-693.21.1.el7.x86_64 and tried compiling master branch again and I was successful. Must have been some dependency page. Last time I yum updated this VM was about a year ago I think.

And... more good news! vector_test.py has been running for a few minutes now and hasn't crashed with malloc error.

Thanks everybody!

closed 2018-05-08 by regulatre