gnuradio/gnuradio #1767: Need help customizing osmocom source and file sink

mehrdaddavoody, 2018-05-08:
Hi

Is there any way to customize an osmocom source which has 4 different central frequencies? And respectively a file sink that has 4 different file saving directories in order for the final flow graph to be smaller in size.

Another question that I have is that can we customize a file sink that is triggered only when a signal is received? (in order to make the saved capture file smaller in size)

marcusmueller, 2018-05-14:
This is a bug tracker, not a help forum; hence, I'm closing this. Please refer to the mailing list for discussion! Thank you very much for your understanding.

closed 2018-05-14 by marcusmueller

dkozel, 2018-05-14:
@mehrdaddavoody The mailing list Marcus is referring to is located at the below link. I hope we can help there.
https://lists.gnu.org/mailman/listinfo/discuss-gnuradio