gnuradio/gnuradio #1769: Socket PDU (server) UDP connection not honoring ICMP port unreachable

regulatre, 2018-05-09:
Given a server with gnuradio socket PDU, listening on a UDP port, a client can connect and get data. But when the client disconnects, the server continues to stream data to the client who is no longer present. So the packets continue to flood the client's interface while the client app has moved on to other things. The Client OS will in turn start to send periodic ICMP unreachable messages to the server to try and tell them to stop. These are basic TCP/IP rules. There's probably some sort of signal or event that the server can catch or listen for to know that the client has disappeared. I'm afraid I'm not a C++ programmer and won't be much help writing a patch, but I wanted to at least call attention to this. 
 
To observe this phenomenon the following may be useful: 

socketPduServer.py sample server: https://gist.github.com/regulatre/7f92815d2d524d2d8776a14c4dd57a3c
For the client, use nc: `$ nc -u -v hostname 5555` (hit control+C to abort the connection). 

All the while monitor the client interface with tcpdump to see the traffic flow `tcpdump -i eth0 -nns0 port 5555 or icmp`

regulatre, 2018-05-09:
This is interesting... if I re-connect the client to the server, the server stops sending to the old dead socket and seems to be resurrected for use but with the new client port.

regulatre, 2018-05-09:
If I make two connections to the server from a single client, the server seems to get confused and only communicate on one of those clients at a time. If one of those clients sends data to the server, the server flips over to that one and if the other one sends something the server flips back to that one. So it appears the server effectively can only talk to a single client at one time, yet it accepts new connections on the socket. I think the socket PDU TCP protocol works with multiple clients so perhaps some of the select/queueing of clients can be ported over to the socket PDU UDP logic.

willcode, 2018-05-09:
Yes, this is how UDP works. You have to put another layer atop UDP to get it to be connection-oriented. Usually, there is another protocol that does setup and then a UDP channel is used as a lower-latency, unreliable alternative to a TCP stream.

noc0lour, 2018-05-09:
@regulatre The UDP server will send packets as long as the OS can route to the destination. To handle ICMP unreachable one would need to implement more logic in the application. 

Generally the zeromq blocks already provide a broadcast mechanism to send data to multiple clients. I'd recommend you try using them for your project.

regulatre, 2018-05-09:
With regard to ICMP unreachable, I'm pretty sure the OS propagates these errors up to the code in the form of an error to send or receive functions that try to use the socket. 

Here are some breadcrumbs I found on Stack Overflow. 
```
Once you connect() a UDP socket, you can use send() and recv() instead of sendto() and recvfrom(). recv() will report an ICMP error for a connected socket.
```
Streaming of data to a client on a closed port should trigger some action to stop the streaming to that port. Certainly it is not normal for a server to stream to a client on a port that has been closed. 

With regard to a single socket being used to service all client requests, imagine if DNS did this - some users would get other users' DNS responses, and other users would never see a response. DNS servers, being UDP servers, have to maintain a table of clients so that it can treat each socket as unique. I believe this is precisely what the Socket PDU block must do in UDP mode. 

Stack Overflow has an example of servicing multiple UDP clients - https://stackoverflow.com/questions/8230876/udp-handling-multiple-clients

Apologies for my lack of C++ knowledge. It's been a while.

noc0lour, 2018-05-09:
@regulatre reading up on some Boost.Asio details:  To send udp packets we are using `send_to()` of the udp socket type. 
This call only throw()s if the endpoint can't be reached (routed) by the OS. Since this is only UDP that's the only thing that matters. The UDP server thus has no way of checking if it should continue/stop streaming to a certain endpoint. Maybe a recvfrom()/recv() on the socket might show errors returned by the OS. 
We would need to implement some logic on the server/client communication to enable/disable endpoints in case of a connect/disconnect. 
The single socket for multiple users should be fine since the socket is called with the `send_to()` method specifying the endpoint we want to send data to. Currently there is only support for one client in the UDP server block (thus only a `d_endpoint_xx` variable).

marcusmueller, 2018-05-14:
@regulatre you should really let go of the "connection" terminology. UDP just sends packets. The sender never hears back from the receiver, and there's no handshaking phase where the existence of any connection is established. So, functionally, I don't see how your proposal maps to UDP?

regulatre, 2018-05-14:
Given a flowgraph acting as a socket PDU server
And given a flowgraph acting as a socket PDU client
And given the client connects to the server and then disconnects - yes you can close a client UDP connection by closing the socket. The OS in turn will generate ICMP unreachables once the socket is closed. 

The observed behavior is that the socket PDU flowgraph (server) will continue to send data to the client at full speed as if the client were still waiting for data. 

A tcpdump of the situation shows the flow of data as expected, but at the moment when the client application terminates, the client socket closes, and the client OS starts to send periodic ICMP unreachables to the server to notify it that the UDP port it is sending to, is not reachable. 

Expected behavior is that the server stops sending data upon receipt of an ICMP unreachable associated with the server socket to which it is sending data. 

Then there's the matter of the server's stated ability being a single connection but the server still accepts multiple connections (letting each successive connection terminate the prior). Should I write that up separately? Expected behavior is that the server ignores inbound packets on the server port if it's already talking to a client.

marcusmueller, 2018-06-21:
> The observed behavior is that the socket PDU flowgraph (server) will continue to send data to the client at full speed as if the client were still waiting for data.

And that is to be expected, and if you do UDP with no flow control of your own on top, absolutely correct.

> A tcpdump of the situation shows the flow of data as expected when the client is active, but at the moment when the client application terminates, the client socket closes(expected), and the client OS starts to send periodic ICMP unreachables(expected) to the server to notify it that the UDP port it is sending to, is not reachable. The server flow graph continues to send packets toward the client (unexpected) without regard to the ICMP messages telling it to stop.

No, that's expected.

By default, a sending UDP socket, simply doesn't *care* about what happens to the packets. As such, it's also not informed by the OS about ICMP packets. Every packet is just sent out in hopes it reaches a target. As if it was the first, or the last one, or the only packet on this planet :)

Now, you can set a DGRAM socket into a connected state, in which in fact `send` calls will fail if the OS got information that the receiving end doesn't accept packets (which is what your ICMP packets do).

Now, we seemingly don't put the socket in that state, if you say you still see packets being sent.

This is correct. These sockets are not meant to be understood as providing a persistent end-to-end connection. It's absolutely legal to use a multicast address as receiver, for example.

> Then there's the matter of the server's stated ability being a single connection but the server still accepts multiple connections (letting each successive connection terminate the prior). Should I write that up separately?

No, it's not a connection. If we state that anywhere in the documentation, we need to change that.

> Expected behavior is that the server ignores inbound packets on the server port if it's already talking to 

No.

Again, this is UDP, if you want a connection concept, use something else.

closed 2018-06-21 by marcusmueller