gnuradio/gnuradio #1770: Socket PDU UDP client doesn't receive data unless it sends data

regulatre, 2018-05-09:
I built a Socket PDU server and client both set to use UDP. 

The server sends a dialtone at 48khz sample rate. https://gist.github.com/regulatre/7f92815d2d524d2d8776a14c4dd57a3c

The client connects to the server, in hopes of receiving the dial tone signal and showing it on a scope sink. 

The client however when executed does not receive any data from the server. After some testing I found that I can get it to receive data if I feed some data into the socket PDU source block. It doesn't seem to care what the data is, and it appears that the connection to the server is not established until at least one byte is sent to the server and then the server sees the client and starts sending to it. 

Expected behavior is that the socket PDU UDP client source block connects to the server and receives data without requiring any input. Perhaps a zero length UDP packet to the server on connect is sufficient.

willcode, 2018-05-09:
True, and only one UDP client can be present at a time, since the server will remember a new UDP client address each time a packet is received from a client. With socket_pdu, client/sink and server/source make sense, but the other way around really doesn't. Since UDP isn't reliable, there's no guarantee that your "startup" packet will get there, so you'd have to have a bunch of handshake logic.

regulatre, 2018-05-09:
Agreed, the server can't send to the client until the client "connects" to the server with at least one packet. From a usability standpoint I find the use of the block difficult sometimes due to "gotchas" like this one. Of course a seasoned network expert knows the ins and outs of UDP but this level of expertise is a tall order for someone wanting to use a block. 

Perhaps requiring the input of the UDP client socket PDU to be connected would be sufficient. 

Or calling out (in the documentation of the block) the fact that data must be sent by the flow graph to initiate the connection. 

The part of me that loves the average user and wants them to succeed tells me it would make the most sense to just take care of it for the user and send a zero-length packet to the server to kick off the data flow so the block works without any gotchas.

willcode, 2018-05-10:
Sending a zero-length packet would just convince the first user who tried to use 2 UDP_CLIENT sources with a UDP_SERVER sink. The problem is pretty specific - UDP_SERVER as a sink for streaming data doesn't work because some higher level protocol is needed. Even sending a startup packet is really a protocol. UDP_CLIENT can work as a source, where another program is already streaming data toward the client. So, requiring the block input to be connected wouldn't help.

I think the right thing is to steer the average use toward ZMQ, as suggested by other here. This block is certainly harder to use with UDP, but would lose flexibility with pretty much any change. Documentation should state that UDP_SERVER sends all packets to the last peer that sent a packet to it.

regulatre, 2018-05-10:
Definitely will be checking out ZMQ. Just wanted to try and help nail down any potential bugs with the Socket PDU blocks before I move on. 

Thank you for taking a look.

closed 2018-05-10 by regulatre