gnuradio/gnuradio #1771: Build failure of the "python3" branch

swap-nil7, 2018-05-11:
While compiling GNU Radio "python3" branch, I am getting an error message regarding the scope of the functions, i.e., the functions `logger_get_level` and `logger_set_level` were not declared in the scope.

Dependencies status: All installed

Command used: `make`

Bash output snippet:
```/home/swapnil/Documents/default/src/gnuradio/gnuradio-runtime/lib/block.cc: In member function _virtual void gr::block::set_log_level(std::__cxx11::string)_:
/home/swapnil/Documents/default/src/gnuradio/gnuradio-runtime/lib/block.cc:720:5: error: _logger_set_level_ was not declared in this scope
     logger_set_level(d_logger, level);
     ^~~~~~~~~~~~~~~~
/home/swapnil/Documents/default/src/gnuradio/gnuradio-runtime/lib/block.cc:720:5: note: suggested alternative: _log_level_
     logger_set_level(d_logger, level);
     ^~~~~~~~~~~~~~~~
     log_level
/home/swapnil/Documents/default/src/gnuradio/gnuradio-runtime/lib/block.cc: In member function _virtual std::__cxx11::string gr::block::log_level()_:
/home/swapnil/Documents/default/src/gnuradio/gnuradio-runtime/lib/block.cc:727:5: error: _logger_get_level_ was not declared in this scope
     logger_get_level(d_logger, level);
     ^~~~~~~~~~~~~~~~
/home/swapnil/Documents/default/src/gnuradio/gnuradio-runtime/lib/block.cc:727:5: note: suggested alternative: _log_level_
     logger_get_level(d_logger, level);
     ^~~~~~~~~~~~~~~~
     log_level
gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:384: recipe for target 'gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/block.cc.o' failed
make[2]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/block.cc.o] Error 1
CMakeFiles/Makefile2:707: recipe for target 'gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/all' failed
make[1]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```
I am facing the same issue while compiling the `next` branch.

fwunsch, 2018-05-14:
Not sure about this, but have you installed log4cpp correctly?

swap-nil7, 2018-05-14:
@fwunsch : log4cpp is found in the gnuradio-runtime dependencies but the according to cmake output, 'liblog4cpp.so.5' may be hidden by files.
Link to the cmake output : [https://pastebin.com/v7EF9VFn](https://pastebin.com/v7EF9VFn)

swap-nil7, 2018-05-16:
Issue Fixed: I did not delete build folder and was running `make` on the same build folder.
So, closing this issue.

closed 2018-05-16 by swap-nil7