gnuradio/gnuradio #1774: Error with "Float to Complex" block if only one input port is connected

om3gafett, 2018-05-14:
I used this sample flowgraph to reproduce:
[image]

I get this error message:
`/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = const void*; _Alloc = std::allocator<const void*>; std::vector<_Tp, _Alloc>::reference = const void*&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.`

If the "Null Source" on the second input of the "Float to Complex" block is enabled, it works without any problems.

I use Fedora 28 and GNU Radio Companion 3.7.11 (installed with dnf)

marcusmueller, 2018-05-14:
Yeah. That's a bug.

Culprit:

      float        *r = (float *)input_items[0];
      float        *i = (float *)input_items[1];

at the beginning of `float_to_complex_impl.cc`'s `work` function.

We need to move this `float *i...` into the `switch(input_items.size())…case 2:` clause.

@om3gafett you wanna do that, or should I?

marcusmueller, 2018-05-14:
@noc0lour whatever F28 does here to make the compiler include these checks at runtime, can we haz that for CI, too?

noc0lour, 2018-05-19:
@marcusmueller can you open an issue on https://github.com/gnuradio/gnuradio-buildbot for that, so I don't forget.

PR #1779 (2018-05-27): move declaration to case
In order to fix  #1774
closed 2018-05-27 by marcusmueller