gnuradio/gnuradio #1783: Problem in code in tutorial

paurea, 2018-05-17:
I am not sure if this is the place for issues in the wiki with tutorials, let me know if it is not. There is a bug in the code for the tutorial. In the code for section 3.1.5 in the code for switching between the two amplitudes  depending on the control signal, the try should include the if above. I have tested it with gnuradio 3.7.11 (included with the latest ubuntu 18).

       def _variable_function_probe_0_probe():
            while True:
                val = self.probe.level()
                print val
                if val == 1:
                    self.set_ampl(1)
                    self.set_freq(1000)
                else:
                    self.set_ampl(.3)
                    self.set_freq(100)
                try:
                    self.set_variable_function_probe_0(val)
                except AttributeError:
                    pass
                time.sleep(1.0 / (10))

should be:

       def _variable_function_probe_0_probe():
            while True:
                val = self.probe.level()
                print val
                try:
                if val == 1:
                    self.set_ampl(1)
                    self.set_freq(1000)
                else:
                    self.set_ampl(.3)
                    self.set_freq(100)
                    self.set_variable_function_probe_0(val)
                except AttributeError:
                    pass
                time.sleep(1.0 / (10))

The problem is the first sample comes before the attribute analog_sig_source_x_1 is created and it is used in set_ampl. The error with the code as it is is:

Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./probep.py", line 92, in _variable_function_probe_0_probe
    self.set_ampl(.3)
  File "./probep.py", line 211, in set_ampl
    self.analog_sig_source_x_1.set_amplitude(self.ampl)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 92, in __getattr__
    return getattr(self._impl, name)
AttributeError: 'top_block_sptr' object has no attribute 'analog_sig_source_x_1'

dkozel, 2018-05-17:
Thanks @paurea. Here or the mailing list is a reasonable place.

mormj, 2019-05-28:
The block_id indeed gets unnecessary quotes.  Not sure yet exactly why this changed from 3.7 to 3.8 - looks like the _stringify_flag for parameters where dtype = string adds additional quotes, which makes sense when the parameter is a function argument, but less sense when the string is the name of a block in the flowgraph. 

Changing the dtype to 'raw' in variable_function_probe yaml will make the issue go away, but not being an expert in the grc codebase i'm not sure if this is the correct fix.  Will have to at a minimum try some others.

There are other 3.7-3.8 conversions missing in gr-tutorial that need to be addressed separately from this bug

PR #2511 (2019-05-29): grc: function_probe_block extra quotes with block_id
The inherent problem is that extra quotes are added when generating the python file - not sure if this is the best solution, but it wasn't clear how to make some string dtypes handled differently than others.  The 'block_id' can be handled as an 'id' dtype because it references another block.  The function name is slightly different, so I added a 'name' dtype with less restrictions than the 'id' dtype.  Please recommend a different solution if this is not preferable.

Due to the way stringify_flag is called, the block_id,
function_name and f [... cut]
commit c8ce648d (2019-05-29): grc: function_probe_block extra quotes with block_id

Due to the way stringify_flag is called, the block_id,
function_name and function_args are all given extra
quotes which cause issues when generating the python source

change/add new dtypes for function_probe

Fixes #1783
closed 2019-05-29 by mbr0wn