gnuradio/gnuradio #1784: make test qa_test_all fails

silverninjastar, 2018-05-17:
I am running Oracle VM VirtualBox with Ubuntu 17.10.1, branch 'maint'
I have been following the [getting started](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux) guide for the USRP B200.  Initially while building, I received an error stating that i needed to use c++11 compiler.  The error stopped the build around 12%.  To remedy this, in CMakeLists.txt I changed line 2 shown from -std=c++98 to -std=c++11 and line 12 shown from -std=gnu99 to -std=gnu11.  This allowed me to complete the build.
```
IF(CMAKE_CXX_COMPILER_ID STREQUAL "GNU")
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
ELSEIF(CMAKE_CXX_COMPILER_ID STREQUAL "Clang")
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++98")
ELSEIF(CMAKE_CXX_COMPILER_ID STREQUAL "MSVC")
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++98")
ELSE()
    message(warning "C++ standard could not be set because compiler is not GNU, Clang or MSVC.")
ENDIF()

IF(CMAKE_C_COMPILER_ID STREQUAL "GNU")
    SET(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=gnu11")
ELSEIF(CMAKE_C_COMPILER_ID STREQUAL "Clang")
    SET(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=gnu99")
ELSEIF(CMAKE_C_COMPILER_ID STREQUAL "MSVC")
    SET(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c11")
ELSE()
    message(warning "C standard could not be set because compiler is not GNU, Clang or MSVC.")
ENDIF()
```
When I ran `make test`, all the tests passed except `qa_test_all`
```
Running tests...
Test project /home/user/workarea-gnuradio/gnuradio/build
    Start 1: qa_test_all
1/7 Test #1: qa_test_all ......................***Failed    1.12 sec
    Start 2: gr-runtime-test
2/7 Test #2: gr-runtime-test ..................   Passed    2.92 sec
    Start 3: gr-pmt-test
3/7 Test #3: gr-pmt-test ......................   Passed    0.83 sec
    Start 4: test_gr_blocks
4/7 Test #4: test_gr_blocks ...................   Passed    0.98 sec
    Start 5: test_gr_filter
5/7 Test #5: test_gr_filter ...................   Passed    0.80 sec
    Start 6: test_gr_analog
6/7 Test #6: test_gr_analog ...................   Passed    0.01 sec
    Start 7: test_gr_digital
7/7 Test #7: test_gr_digital ..................   Passed    0.05 sec

86% tests passed, 1 tests failed out of 7

Total Test time (real) =   6.72 sec

The following tests FAILED:
	  1 - qa_test_all (Failed)
Errors while running CTest
Makefile:129: recipe for target 'test' failed
make: *** [test] Error 8
```

noc0lour, 2018-05-17:
`qa_test_all` is inherited from the volk submodule. If you run `ctest --output-on-failure` or `ctest -V` you will get more information about which test fails.

silverninjastar, 2018-05-17:
I executed the `ctest -V` command and found the following info.
```
RUN_VOLK_TESTS: volk_32f_8u_polarbutterflypuppet_32f(131071,1)
1: generic completed in 9.539ms
1: u_avx completed in 0.874ms
1: offset 16 in1: 2 in2: 0 tolerance was: 1e-06
1: offset 17 in1: 4 in2: -2 tolerance was: 1e-06
1: offset 18 in1: -4 in2: -0 tolerance was: 1e-06
1: offset 19 in1: 8 in2: 4 tolerance was: 1e-06
1: offset 20 in1: -4 in2: -0 tolerance was: 1e-06
1: offset 21 in1: 8 in2: 4 tolerance was: 1e-06
1: offset 22 in1: 8 in2: -0 tolerance was: 1e-06
1: offset 23 in1: -16 in2: 8 tolerance was: 1e-06
1: offset 24 in1: -4 in2: 0 tolerance was: 1e-06
1: offset 25 in1: -8 in2: -4 tolerance was: 1e-06
1: volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx
1: offset 16 in1: 0 in2: 1 tolerance was: 0
1: offset 17 in1: 0 in2: 1 tolerance was: 0
1: offset 22 in1: 0 in2: 1 tolerance was: 0
1: offset 23 in1: 1 in2: 0 tolerance was: 0
1: offset 33 in1: 0 in2: 1 tolerance was: 0
1: offset 35 in1: 0 in2: 1 tolerance was: 0
1: offset 36 in1: 0 in2: 1 tolerance was: 0
1: offset 37 in1: 0 in2: 1 tolerance was: 0
1: offset 38 in1: 0 in2: 1 tolerance was: 0
1: offset 39 in1: 1 in2: 0 tolerance was: 0
1: volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx
1: Best aligned arch: generic
1: Best unaligned arch: generic
1: Failure on volk_32f_8u_polarbutterflypuppet_32f
1: Kernel QA finished: 1 failures out of 101 tests.
```

noc0lour, 2018-05-19:
try checking out the latest release or master from https://github.com/gnuradio/volk.git separately and build & test it. If the error persists there is still a bug

dkozel, 2018-06-17:
@silverninjastar Could you please confirm if you still see this issue?

closed 2018-12-13 by mbr0wn