gnuradio/gnuradio #1798: Missing arguments in calls to MessageDialogHelper

simark, 2018-05-29:
I got this call stack

```
Traceback (most recent call last):
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\gui\ActionHandler.py", line 523, in _handle_action
    file_paths = args if args else OpenFlowGraphFileDialog(main, page.get_file_path()).run()
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\gui\FileDialogs.py", line 204, in run
    filename = self.get_rectified_filename()
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\gui\FileDialogs.py", line 192, in get_rectified_filename
    Utils.parse_template(FILE_DNE_MARKUP_TMPL, filename=filename),
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\gui\Dialogs.py", line 151, in MessageDialogHelper
    message_dialog = gtk.MessageDialog(parent, gtk.DIALOG_MODAL, type, buttons)
TypeError: parent should be a GtkWindow or None
```

Some calls to MessageDialogHelper seem to be missing the parent parameter.

dkozel, 2018-06-17:
@simark Thanks for the report, sorry it's taken so long to get around to it.

Do you have a reliable way to trigger that warning? Do you know what release of GNU Radio you're running?

simark, 2018-06-17:
I don't have access to the file that lead to this in the first place.  The situation was a file created on Linux (with a name possibly containing spaces and accented characters), and I was trying to open it on Windows.

But it should be possible to simulate this situation using master by patching the condition that leads to MessageDialogHelper being called:

```diff
diff --git a/grc/gui/FileDialogs.py b/grc/gui/FileDialogs.py
index 774a76714cde..1e4434a182a6 100644
--- a/grc/gui/FileDialogs.py
+++ b/grc/gui/FileDialogs.py
@@ -186,7 +186,7 @@ class FileDialog(FileDialogHelper):
         elif self.type in (OPEN_FLOW_GRAPH, OPEN_QSS_THEME):
             filenames = self.get_filenames()
             for filename in filenames:
-                if not path.exists(filename): #show a warning and re-run
+                if True: #show a warning and re-run
                     MessageDialogHelper(
                         gtk.MESSAGE_WARNING, gtk.BUTTONS_CLOSE, 'Cannot Open!',
                         Utils.parse_template(FILE_DNE_MARKUP_TMPL, filename=filename),
```

luzpaz, 2023-07-22:
Is this still relevant ? (I can't find it in https://github.com/gnuradio/gnuradio/blob/main/grc/gui/FileDialogs.py)

marcusmueller, 2023-07-22:
@luzpaz according to the Windows paths, this is 3.7-era code. 

@simark I think this got fixed a while ago, and if it's only present in 3.7, we couldn't fix it anymore, anyways. If you still are facing this issue with a modern GNU Radio, please ping me, to reopen the issue. Thanks!

closed 2023-07-22 by marcusmueller