gnuradio/gnuradio #1801: digital: Add Constellation Encoder implementation

noc0lour, 2018-05-30:
Currently the only way to convert bytes to symbols using a constellation object is either using the `chunks_to_symbols` or the `constellation_modulator`. 

The former is not intuitive and won't work for differential phase modulation. The latter is applying pulse shaping which might not be desired (and is a python hier block).

A simple Constellation Encoder (the inverse to the Constellation Decoder) might be worth the effort. 
I can implement that one and deprecate the others.  According to: https://docs.google.com/spreadsheets/d/1Yty89iTZbr1zM_2M8NyypdNz-P5TN2IvgXApDf94rsk/edit#gid=0

marcusmueller, 2018-08-18:
@noc0lour keep this as a 3.8 milestone or postpone to 3.9?

noc0lour, 2018-08-18:
This should be doable in a afternoon session, since the way is now rather paved to the 3.8 release I could throw that in, if you don't mind?

PR #3533 (2020-11-16): digital: add constellation_encoder block
Add a new block that is the reverse of constellation decoder, to allow more flexibility in making modulation flowgraphs (not requiring chunks to symbols, constellation modulator, etc.)

[image]

Fixes #1801
closed 2020-11-16 by mormj