gnuradio/gnuradio #1805: Files require C++11, but that isn't enabled by the compiler flags

MattEttus, 2018-05-31:
Compiling on fresh Fedora 26 machine....

[ 54%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/float_array_to_int.cc.o
In file included from /usr/include/c++/7/cstdint:35:0,
                 from /home/parallels/gnuradio/gr-blocks/lib/float_array_to_int.cc:30:
/usr/include/c++/7/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^~~~~
/home/parallels/gnuradio/gr-blocks/lib/float_array_to_int.cc:32:33: error: ‘INT32_MAX’ was not declared in this scope
 static const int64_t MAX_INT =  INT32_MAX;
                                 ^~~~~~~~~
/home/parallels/gnuradio/gr-blocks/lib/float_array_to_int.cc:32:33: note: suggested alternative: ‘__INT32_MAX__’
 static const int64_t MAX_INT =  INT32_MAX;
                                 ^~~~~~~~~
                                 __INT32_MAX__
/home/parallels/gnuradio/gr-blocks/lib/float_array_to_int.cc:33:33: error: ‘INT32_MIN’ was not declared in this scope
 static const int64_t MIN_INT =  INT32_MIN;
                                 ^~~~~~~~~
make[2]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/build.make:3019: gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/float_array_to_int.cc.o] Error 1

lukaskuzmiak, 2018-06-01:
yep, got the same thing with v3.7.13.2 on Ubuntu 16.04 just now

noc0lour, 2018-06-01:
Oh, seems like @marcusmueller pulled those changes into the 3.7-maint release... 
This should float on `<stdint.h>` instead of `<cstdint>`

curtcorum, 2018-06-09:
https://github.com/gnuradio/gnuradio/pull/1810

marcusmueller, 2018-06-09:
Closed by merging #1810.

closed 2018-06-09 by marcusmueller