gnuradio/gnuradio #1811: python crashing upon executing a flowgraph

karanjarich, 2018-06-05:
Hey  Guys,
I have been trying to solve this issue but in vain. I cloned Gnu Radion Companion example in order to learn Digital signal processing, I have managed to import them to GRC, unfortunately, once  I click execute button to be able to view the results the program is halted giving an error message that python.exe has stopped working what could be the problem I'm using windows 7. Thanks

dkozel, 2018-06-17:
Hello @karanjarich 

I'm afraid we can't even start to guess without knowing more. What version of GNU Radio are you running and what examples are you trying to run?

karanjarich, 2018-06-18:
thanks for you concern currently am using gnu radio companion 3.7.11

aka7acharya, 2019-01-08:
hey, i had the same issue and i use the same version of gnu radio companion. was the issue solved? if so how?

gnieboer, 2019-02-13:
What method do you use to installer gnuradio on windows?  PothosSDR or the installers from gcndevelopment?  

Have you tried using newer versions?  

Can you give us a screenshot of the python console command prompt behind the GRC GUI?   Before you dis issued the python error.  That will show us more information about the error

NeuroForLunch, 2021-06-24:
Stale issue and more clarification not provided. Please close

marcusmueller, 2023-12-21:
3.7 era

closed 2023-12-21 by marcusmueller