gnuradio/gnuradio #1840: Investigate failing `qa_constellation_soft_decoder_cf` tests

marcusmueller, 2018-06-23:
@noc0lour any idea?

noc0lour, 2018-06-24:
using python randomness without a seed might lead to results slightly off. also the number of samples (100) isn't really "large".
Probably this unit test should be replaced with a reproducable one (with a seeded RNG).

noc0lour, 2018-12-12:
```FAIL: test_constellation_soft_decoder_cf_bpsk_8_rand (__main__.test_constellation_soft_decoder)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/cache/workdir/Ubuntu_16_04_64_0/build_Ubuntu_16_04_64/src/gr-digital/python/digital/qa_constellation_soft_decoder_cf.py", line 105, in test_constellation_soft_decoder_cf_bpsk_8_rand
    self.helper_with_lut(prec, src_data, digital.psk_2_0x0, digital.sd_psk_2_0x0)
  File "/cache/workdir/Ubuntu_16_04_64_0/build_Ubuntu_16_04_64/src/gr-digital/python/digital/qa_constellation_soft_decoder_cf.py", line 58, in helper_with_lut
    self.assertFloatTuplesAlmostEqual(expected_result, actual_result, 5)
  File "/cache/workdir/Ubuntu_16_04_64_0/build_Ubuntu_16_04_64/src/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 102, in assertFloatTuplesAlmostEqual
    self.assertAlmostEqual (a[i], b[i], places, msg)
AssertionError: 0.96078431372549011 != 0.9686274528503418 within 5 places
```

noc0lour, 2018-12-12:
https://ci.gnuradio.org/buildbot/#/builders/49/builds/370/steps/7/logs/stdio

michaelld, 2018-12-12:
Maybe due to https://github.com/gnuradio/gnuradio/commit/3359d67e02158199d24756155fb58ab43b7b66ef ?

marcusmueller, 2021-03-24:
cannot reproduce

closed 2021-03-24 by marcusmueller