gnuradio/gnuradio #1841: remove non-thread-safe `rand` in remaining code

marcusmueller, 2018-06-23:
Much as in the spirit of https://github.com/gnuradio/gnuradio/pull/1562, we need to *completely* eradicate `rand` from our code base wherever things  *could* be multithreaded. That's basically everywhere, even, and especially, in test cases.

marcusmueller, 2018-06-23:
This will entail replacing `rand`, `srand` and friends with `xoroshiro128p_next` in a lot of places.

marcusmueller, 2019-04-19:
Seems like it's done.

rear1019, 2019-05-25:
> Seems like it's done.

I beg to differ. See output of `git grep -E "\<s?rand\>[ \t]*\(" -- "*.cc" "*.c" "*.h"`:

- `gr-digital/lib/ofdm_mapper_bcv_impl.cc`
- `gr-digital/lib/qa_header_format.cc` (A fix can be backported from master if I remember correctly.)
- `gr-trellis/lib/interleaver.cc`
- three files in `gr-vocoder/lib/codec2`
- `cmake/msvc/config.h`

The last file provides a wrapper for `[s]random()` (defined by POSIX) which calls `[s]rand()`. `[s]random()` is used by QA code (see output of `git grep -E "\<s?random\>[ \t]*\(" -- "*.cc" "*.c" "*.h"`).

rear1019, 2019-06-11:
Please assign to me, I will look into the remaining usage of `[s]rand()` and `[s]random()`.

rear1019, 2019-06-12:
Pull request to remove `[s]random()` #2549. Concerning the usage of `[s]rand()`:

- `gr-digital/lib/ofdm_mapper_bcv_impl.cc`: Changes to code might be sort of an API change as user visible behavior changes. How should I proceed here?
- `gr-trellis/lib/interleaver.cc`: As above. This interleaver is used by gr-trellis only as far as I can tell.
- `gr-digital/lib/qa_header_format.cc`: Fix can be back-ported from master.
- `gr-vocoder/lib/codec2`: Files using `rand()` are not compiled or used in gr-vocoder. No changes are required here IMO.

marcusmueller, 2019-06-16:
Thank you for this!

The API change thing is hard. We officially can't do it on 3.7.x.y, because API is signified by the 7. We can't fix things and release them as 3.8, because that's a different thing already.

However, looking back at the history of the 3.7 releases, I'm afraid we won't significantly make matters less consistent than they are.

So, yeah, we can simply tag this with "ABI break" on github, and say "interface change" in the commit msg, so that I don't forget to increase the third versioning place for the next maint-3.7 release.

marcusmueller, 2021-03-22:
Done! Thanks.

closed 2021-03-22 by marcusmueller